blueman-project/blueman #2380: ConnectionNotifier configuration

AriaMoradi, 2024-05-21:
My mouse has a standby mode that makes it turn off to save battery, this makes it connect and disconnect periodically and thus generates really annoying notification, It seems that the ConnectionNotifier plugin doesn't have any configuration options, having an option to disable notifications for a specific device will go a long way.

blueman: 2.4.1
BlueZ: 5.70 (Downgraded manually)
Distribution: Arch Linux
Desktop environment: Sway

cschramm, 2024-05-21:
A notification every few seconds means that the device goes to sleep pretty much immediately whenever there is no movement? What model is that?

AriaMoradi, 2024-05-22:
> A notification every few seconds means that the device goes to sleep pretty much immediately whenever there is no movement? What model is that?

Its called Xiaomi Mi Silent Mouse, product number is WXSMSBMW02.

igorsantos07, 2024-08-22:
A way to disable either connection or disconnection alerts would be useful as well. I don't want to be notified when my keyboard and mouse connect - as it's easy to check that, but I do want to know if they suddenly go off.

rohoog, 2024-09-14:
I think this should be some kind of extra attribute for 'trusting' the device. Like 'trust and mute connection notifications'. Things you don't trust must certainly still produce such notifications. Alternatively, a filter setting for the ConnectionNotifier to mute notifications for certain classes of devices (that are trusted as well)..

shawnyeager, 2025-01-15:
The chatter of notifications is something I'd really, really like to be able to manage or outright silence.