blueman-project/blueman #2383: `AudioProfiles`'s item isn't a sub-menu on auto-connection

kuraga, 2024-05-22:
blueman: 2.4.1
BlueZ: 5.75
Distribution: Calculate Linux (Gentoo-based)
Desktop environment: XFCE 4.18

* [ x ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

In most cases,
on auto-connection initiated by headphones (https://github.com/blueman-project/blueman/issues/1709#issuecomment-1115324427),
the sub-menu-item by `AudioProfiles` (applet) plugin
is just a menu-item (i.e. _it isn't a sub-menu_).
It's exactly not-sub-menu. It's _not_ an empty sub-menu.

I don't think it's a serious issue but probably this observation could help in other cases.

kuraga, 2024-05-22:
On my observation, `items` [here](https://github.com/blueman-project/blueman/blob/2.4.1/blueman/plugins/applet/PulseAudioProfile.py#L79) is legal, i.e. the standard list of sub-items-data.

cschramm, 2024-05-23:
You see an `Audio Profiles for ...` item without a sub menu and that's 100 % reproducible when the device initiates the connection but 0 % when you do from the system that's running blueman? You see sub menu items getting generated in the script, though?

Does restarting `blueman-tray` fix it (`killall blueman-tray`, then run it manually)?

Please provide `dbus-send --print-reply --dest=org.blueman.Applet /org/blueman/Applet org.blueman.Applet.GetMenu` and `blueman-tray --loglevel debug` (with no `blueman-tray` process running).

kuraga, 2024-05-23:
> You see an `Audio Profiles for ...` item without a sub menu and that's 100 % reproducible when the device initiates the connection but 0 % when you do from the system that's running blueman?

Well, 80-90%.

> You see sub menu items getting generated in the script, though?

Yes.

> Does restarting `blueman-tray` fix it (`killall blueman-tray`, then run it manually)?

No.

> Please provide `dbus-send --print-reply --dest=org.blueman.Applet /org/blueman/Applet org.blueman.Applet.GetMenu`

(at  the moment of the described situation - i.e. item instead of item-with-sub-menu)
https://gist.github.com/kuraga/0ae5bb9d2018666108aa0514ec04d370

> and `blueman-tray --loglevel debug` (with no `blueman-tray` process running).

```
$ blueman-tray --loglevel=DEBUG
(blueman-tray:28259): dbind-WARNING **: 18:50:44.592: Couldn't connect to accessibility bus: Failed to connect to socket /root/.cache/at-spi/bus_0: Отказано в доступе
blueman-tray 18.50.44 DEBUG    Tray:36 _on_name_appeared: Applet started on name org.blueman.Applet, showing indicator
blueman-tray 18.50.44 INFO     Tray:46 _on_name_appeared: Using indicator "StatusNotifierItem"
```

(no output on auto-connection)

kuraga, 2024-05-23:
Interesting, there is the sub-menu in the manager window, at the same moment.

cschramm, 2024-05-24:
So `blueman-applet` provides a proper sub menu to `blueman-tray` and `blueman-tray` uses the StatusNotifierItem implementation. Any data from the tray, e.g. with `xfce4-panel --restart`?

kuraga, 2024-05-24:
Now caught some different case, _no menu item at all_:

[image]

And `xfce4-panel --restart` helped!

[image]

kuraga, 2024-05-24:
Is there a race condition?.. The callback doesn't try to convert to a sub-menu since something around isn't ready yet...

kuraga, 2024-05-26:
BTW, on auto-connection, the item appears _twice_. And it doesn't have the sub-menu on the second time.

Also now I see the issue rarely. Don't know have been changed.

closed 2024-08-04 by github-actions[bot]