blueman-project/blueman #2388: Blueman tray icon context menu doesnt appear

alejomka, 2024-05-23:
blueman: 2.3.5
BlueZ: 5.64-0ubuntu1.1
Distribution: Mint 21.3
Desktop environment: Xfce 4.18

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

When i right click on the bluetooth tray icon doesn’t show blueman context menu, the “Status Tray Plugin” menu appears instead.
Left click works fine and opens the bluetooth device window.
I have also tried changing the panel settings so that left clicking opens the context menu, but when I do that and touch the bluetooth icon, it does absolutely nothing

Already tried to reinstall blueman, purge, reboot: no effect.

cschramm, 2024-06-16:
Please `killall blueman-tray` and run `blueman-tray --loglevel debug` to double check the implementation, but I assume `StatusNotifierItem`. Does `xfce4-panel` log something e.g. to `~/.xsession-errors`?

PeterDaveHello, 2024-08-16:
Please keep it open, thanks.

PeterDaveHello, 2024-10-17:
This issue has not been resolved and should not be automatically closed. It requires further investigation and a proper fix. Thank you.

cschramm, 2024-10-17:
This lacks information for four months now. You're welcome to contribute.

infirit, 2024-10-17:
There is a reopen button which I'll click when the information is provided.

closed 2024-10-17 by infirit