blueman-project/blueman #2411: Feature/improvement: taskbar icon when using dark theme

jayjayseal, 2024-06-19:
blueman-(manager):
BlueZ:
Distribution: Linux Mint Cinnamon
Desktop environment: 

Feature/improvement

Hi,

I'm using the blueman-(manager) and i noticed that when you are using the dark theme the icon in the taskbar becomes almost invisible when there are no devices connected. Is it possible to change the icon to something else? In the past this was a nice B/W icon with a white dot to show if a device was connected.

[image]

admittedly this would also be nicer against my own current icons

cschramm, 2024-06-19:
That does not look like any of our icons.

infirit, 2024-06-19:
Does `gsettings set org.blueman.general symbolic-status-icons true` work for you?

jayjayseal, 2024-06-20:
> Does `gsettings set org.blueman.general symbolic-status-icons true` work for you?

Yes, this worked :-) Thank you! :)

closed 2024-08-27 by github-actions[bot]