blueman-project/blueman #2423: Problema com Bluetooth

emerson-molusco, 2024-07-01:
blueman: 
BlueZ:
Distribution: Endeavour 
[image]

Desktop environment: Cinnamon

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2024-08-10:
`systemctl status bluetooth.service`

emerson-molusco, 2024-08-13:
[image]

> `systemctl status bluetooth.service`

deu isso.

cschramm, 2024-08-13:
Well, from what I see it does not seem to be running.

emerson-molusco, 2024-08-14:
what i have to do??