blueman-project/blueman #2429: Gtk-WARNING when device name contains "markup"

XeCycle, 2024-07-15:
blueman: 2.4.2
BlueZ: 5.77
Distribution: Arch Linux
Desktop environment: awesome wm, notification daemon dunst

* [*] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

My headset, Creative Zen Hybrid Pro, reports a weird name.  As an example [this reporter](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/3666) says their device name is "ZenHybridPro (SX:A0l9g7[m[`)"; mine is "ZenHybridPro (SX:p<>v......)" .I'm not sure whether this name can be linked to MAC or whatever, allow me to mask part of it).  And blueman-applet frequently logs

> (blueman-tray:15759): Gtk-WARNING **: 10:44:17.326: Failed to set text '音频和输入配置文件 on ZenHybridPro (SX:p<>v......)' from markup due to error parsing markup: Error on line 1 char 59: “>” is not a valid character following a “<” character; it may not begin an element name

The Chinese part means "audio and input profile".  This is shown 3 times at applet startup, and also several times when device connects.  I tried `--loglevel debug` but no relevant logs for this one.

infirit, 2024-07-15:
Ugh, looks like we can't trust the name we get from BlueZ. I suspect the error is from here. https://github.com/blueman-project/blueman/blob/42110d9c305868f853a455ecdb821a120d560fb9/blueman/main/indicators/StatusNotifierItem.py#L75

infirit, 2024-07-15:
Can you try below debugging (and possible fix) patch. It also logs the the device name (alias) to the applet log so there may be something interesting in `blueman-applet --loglevel debug`.

```patch
diff --git a/blueman/plugins/applet/RecentConns.py b/blueman/plugins/applet/RecentConns.py
index 84ec37a07..28e825da5 100644
--- a/blueman/plugins/applet/RecentConns.py
+++ b/blueman/plugins/applet/RecentConns.py
@@ -1,5 +1,6 @@
 from gettext import gettext as _
 from operator import itemgetter
+import html
 import time
 import logging
 from typing import List, TYPE_CHECKING, Optional, Callable, cast, Union
@@ -167,8 +168,10 @@ class RecentConns(AppletPlugin, PowerStateListener):
         self.parent.Plugins.DBusService.connect_service(item["device"], item["uuid"], reply, err)
 
     def _build_menu_item(self, item: "Item") -> "SubmenuItemDict":
+        logging.debug(f"{item['name']} - {item['alias']}")
+        alias = html.escape(item["alias"])
         mitem: "SubmenuItemDict" = {
-            "text": _("%(service)s on %(device)s") % {"service": item["name"], "device": item["alias"]},
+            "text": _("%(service)s on %(device)s") % {"service": item["name"], "device": alias},
             "markup": True,
             "icon_name": item["mitem"]["icon_name"] if item["mitem"] is not None else item["icon"],
             "sensitive": item["device"] is not None,
```

XeCycle, 2024-07-16:
@infirit Yes that fixes it!

PR #2431 (2024-07-17): RecentConns: Alias may contain < > which causes parse errors
For example in the StatusNotifierItem.

fixes #2429
commit 77bfb451 (2024-07-17): RecentConns: Alias may contain < > which causes parse errors

For example in the StatusNotifierItem.

fixes #2429
PR #2432 (2024-07-24): RecentConns: Alias may contain < > which causes parse errors
For example in the StatusNotifierItem.

fixes #2429
commit 92cc6fcd (2024-07-24): RecentConns: Alias may contain < > which causes parse errors

For example in the StatusNotifierItem.

fixes #2429
closed 2024-07-17 by infirit