blueman-project/blueman #2436: Rii Blutooth and 2.4G Mini Keyboard

LAP47, 2024-07-20:
blueman:
BlueZ:
Distribution: Linux Mint 22 Beta
Desktop environment: Mate'

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Blueman doesn't find the keyboard when searching for it so the keyboard only works when using the included 2.4G USB Adapter which I'm using right now.

cschramm, 2024-08-10:
Did you try `bluetoothctl`? Is it discoverable from another device at the same time?

LAP47, 2024-08-10:
> Did you try `bluetoothctl`? Is it discoverable from another device at the same time?

No. I haven't. I just went right to using the 2.4GHz once I decided that I wasn't able to figure it out.

closed 2024-10-17 by github-actions[bot]