blueman-project/blueman #2448: Impossible to pair a bluetooth mouse via keyboard shortcuts

nickspiker, 2024-08-05:
You can turn on/off and scan for bluetooth devices, you can even select the mouse that shows up in the list but enter and specebar do nothing.

infirit, 2024-08-05:
You can pair/connect from the device menu. Is that menu not available for you?

nickspiker, 2024-08-05:
Nope.  I can see the menu but tab only goes back and fourth between the top buttons and the devices so I can't actually do anything with the menus.  When you right click the menu comes up (I had to pair thru bash as I didn't have a mouse to right click of course)

infirit, 2024-08-05:
Ah, I see you haven't been introduced to accelerators on menus. Hold `Alt` and one of the letters in the menu will be underlined.  Then while holding alt press the underlined key (d in my example) on your keyboard. Like this.
[image]

infirit, 2024-08-05:
Ooh, but I see the bug now. Let me check that out. In the mean time use the menu with the accelerator.

infirit, 2024-08-05:
Not having a great day. Here is how it should work.

* Tab moves between sections
* When in the toolbar use the left and right arrow keys to switch between the buttons.
* Buttons that are not sensitive (for example the device is already paired) are not accessible.

The issue I think there is no visible indication that the button is not sensitive as the icon doesn't change.

infirit, 2024-08-05:
@nickspiker Would the top screenshot in #2449 help you better identifying when a button can't be pressed?

nickspiker, 2024-08-05:
Ah, you have to hold alt, yes.  I tried that but I didn't 'Hold' it.  Typically alt brings you to the menus so you can arrow thru them but I think the biggest thing is that enter and space should bring up the menu.  That way you can choose what to do with the device if you don't have a mouse and I intuitively didn't even think to hold alt so...

nickspiker, 2024-08-05:
Checking other apps and yes, holding alt seems to be the default so I think just the menu when selected is the one that should be added.

closed 2024-10-13 by github-actions[bot]