blueman-project/blueman #2451: "Info" values are difficult to parse, visually and programmatically.

RokeJulianLockhart, 2024-08-07:
#### **Prerequisites**

<table><tr>
	<th>Blueman</th>
	<td></td>
</tr><tr>
	<th>BlueZ</th>
	<td></td>
</tr><tr>
	<th>Distribution</th>
	<td></td>
</tr><tr>
	<th>Desktop environment</th>
	<td></td>
</tr></table>

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting/b4ae37f563a62ecaa07d6a9a631b681cb82836ff#troubleshooting-blueman) and done my best effort to follow.

#### **Request**

1.	When I duplicate the content of the undermentioned window to my clipboard:

	> [image]

1.	...the result is nigh incomprehensible:

	<blockQuote>

	```txt
	88:C9:E8:3A:02:CE
	public
	WH-1000XM5
	WH-1000XM5
	0x240404
	0x0000
	audio-headset
	yes
	yes
	no
	no
	no
	00000000-deca-fade-deca-deafdecacaff Proprietary
	00001108-0000-1000-8000-00805f9b34fb Headset
	0000110b-0000-1000-8000-00805f9b34fb Audio Sink
	0000110c-0000-1000-8000-00805f9b34fb Remote Control Target
	0000110d-0000-1000-8000-00805f9b34fb Advanced Audio
	0000110e-0000-1000-8000-00805f9b34fb Remote Control
	0000110f-0000-1000-8000-00805f9b34fb Video Conferencing
	0000111e-0000-1000-8000-00805f9b34fb Handsfree
	00001124-0000-1000-8000-00805f9b34fb Human Interface Device Service (HID)
	00001131-0000-1000-8000-00805f9b34fb Headset
	00001200-0000-1000-8000-00805f9b34fb PnP Information
	45c93e07-d90d-4b93-a9db-91e5dd734e35 Proprietary
	764cbf0d-bbcb-438f-a8bb-6b92759d6053 Proprietary
	81c2e72a-0591-443e-a1ff-05f988593351 Proprietary
	8901dfa8-5c7e-4d8f-9f0c-c2b70683f5f0 Proprietary
	931c7e8a-540f-4686-b798-e8df0a2ad9f7 Proprietary
	956c7b26-d49a-4ba8-b03f-b17d393cb6e2 Proprietary
	9b26d8c0-a8ed-440b-95b0-c4714a518bcc Proprietary
	df21fe2c-2515-4fdb-8886-f12c4d67927c Proprietary
	f76acb00-7cab-495f-bb1a-e664598fd77f Proprietary
	f7a96061-a1b3-40de-aff0-e78ec45a151e Proprietary
	f8d1fbe4-7966-4334-8024-ff96c9330e15 Proprietary
	usb:v054Cp0DF0d0210
	/org/bluez/hci0
	```

	</blockQuote>

Consequently, I suggest that you instead copy those values in some kind of key-value pair specification, like JSON (or XML).

RokeJulianLockhart, 2024-08-07:
> My original use case was simply that I wanted the values to paste in other places for development and testing. Having it in XML/JSON would make that a lot more annoying.

@infirit, would https://en.wikipedia.org/w/index.php?title=INI_file&oldid=1232644417#Example suit you better, then? That's what FreeDesktop uses.

> Can you provide your use case and why you need it in another format? And do you really need all those values? Typically you would only want a couple, like `btaddress`, and maybe the icon name.

I use it to compare hardware information across devices.

infirit, 2024-08-08:
The device info window/plugin is for simple viewing and retrieving some values. I personally like to keep it that way.

If you want parse-able data get the properties directly from bluez over dbus with something like below. 

```
gdbus call --system --dest org.bluez --object-path /org/bluez/hci0/dev_00_16_94_38_38_F2 --method org.freedesktop.DBus.Properties.GetAll org.bluez.Device1
```
Which outputs
```
({'Address': <'00:16:94:38:38:F2'>, 'AddressType': <'public'>, 'Name': <'HD 4.40BT'>, 'Alias': <'HD 4.40BT'>, 'Class': <uint32 2360324>, 'Icon': <'audio-headset'>, 'Paired': <true>, 'Bonded': <true>, 'Trusted': <true>, 'Blocked': <false>, 'LegacyPairing': <false>, 'Connected': <false>, 'UUIDs': <['00001108-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000110d-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '0000110f-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb', '00001131-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb']>, 'Modalias': <'bluetooth:v0A12p0001d0000'>, 'Adapter': <objectpath '/org/bluez/hci0'>, 'ServicesResolved': <false>},)
```

GLib/Gio provides nice DBus classes that hides a lot of the details from the user and has bindings for many languages.

RokeJulianLockhart, 2024-08-08:
> The device info window/plugin is for simple viewing and retrieving some values. I personally like to keep it that way.

@infirit, could the value and key at least be on the same line then, separated by something like a tab? That would then give me an easily readable TSV. I can't imagine you find it easier to read each value on a new line?

> If you want parse-able data get the properties directly from bluez over dbus.

Thanks. I'll do that.