blueman-project/blueman #2458: Blueman suddently stopeed detecting my devices.

Scottie5000, 2024-08-13:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

 Suddenly Blueman cannot detect ANY of my bluetooth devices. Did so just fine with 21.3 and prior versions. Worked initially on 22, and did some updates today 08.12.2024 and it still worked, rebooted my pc and it no longer detects anything. Help

Apple mouse, apple keyboard, sonos move

infirit, 2024-08-13:
Probably better to ask on the Mint forums as it happened after updating something.

There may be some hints in the blueman-applet debug log. See https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman

Scottie5000, 2024-08-13:
NM upon reboot it's working just fine. Just a glitch in the Matrix.

closed 2024-08-13 by infirit