blueman-project/blueman #2465: blueman manager pair

antonvdh, 2024-08-18:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

antonvdh, 2024-08-18:
I wanted to pair my bluetooth keyboard but I did not manage because the pincode will only  be visible for 3 seconds

PapaJS, 2024-08-18:
Connects with my bluetooth keyboard but will disconnect after about 8 sec. Won't stay connected.

infirit, 2024-08-22:
> I wanted to pair my bluetooth keyboard but I did not manage because the pincode will only be visible for 3 seconds

Fill in the issue template and post a blueman-applet debug log of you trying to pair. See how to do the debug log here: https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman

closed 2024-09-06 by github-actions[bot]