blueman-project/blueman #2479: Provide a monochromatic icon for the system tray.

RokeJulianLockhart, 2024-09-02:
#### **Prerequisites**

1.	#### **Versions**

	<table><tr>
		<th>Blueman</th>
		<td> 
			<table><tr>
				<th>Version</th>
				<td><a href="https://kojipkgs.fedoraproject.org//packages/blueman/2.4.3/1.fc40/x86_64/blueman-2.4.3-1.fc40.x86_64.rpm">https://kojipkgs.fedoraproject.org//packages/blueman/2.4.3/1.fc40/x86_64/blueman-2.4.3-1.fc40.x86_64.rpm</a></td>
			</tr><tr>
				<th>Source</th>
				<td><a href="https://koji.fedoraproject.org/koji/rpminfo?rpmID=39543260">https://koji.fedoraproject.org/koji/rpminfo?rpmID=39543260</a></td>
			</tr></table>
		</td>
	</tr><tr>
		<th>BlueZ</th>
		<td> 
			<table><tr>
				<th>Version</th>
				<td><a href="https://kojipkgs.fedoraproject.org//packages/bluez/5.77/1.fc40/x86_64/bluez-5.77-1.fc40.x86_64.rpm">https://kojipkgs.fedoraproject.org//packages/bluez/5.77/1.fc40/x86_64/bluez-5.77-1.fc40.x86_64.rpm</a></td>
			</tr><tr>
				<th>Source</th>
				<td><a href="https://koji.fedoraproject.org/koji/rpminfo?rpmID=39114955">https://koji.fedoraproject.org/koji/rpminfo?rpmID=39114955</a></td>
			</tr></table>
		</td>
	</tr><tr>
		<th>Distribution</th>
		<td>
			<table><tr>
				<th>Version</th>
				<td><code><a href="cpe:/o:fedoraproject:fedora:40">cpe:/o:fedoraproject:fedora:40</a></code></td>
			</tr><tr>
				<th>Source</th>
				<td><a href="https://download.fedoraproject.org/pub/fedora/linux/releases/40/Spins/x86_64/iso/Fedora-KDE-Live-x86_64-40-1.14.iso">https://download.fedoraproject.org/pub/fedora/linux/releases/40/Spins/x86_64/iso/Fedora-KDE-Live-x86_64-40-1.14.iso</a></td>
			</tr></table>
		</td>
	</tr><tr>
		<th>Desktop environment</th>
		<td>
			<table><tr>
				<th>Version</th>
				<td><a href="https://kojipkgs.fedoraproject.org//packages/plasma-desktop/6.1.3/2.fc40/x86_64/plasma-desktop-6.1.3-2.fc40.x86_64.rpm">https://kojipkgs.fedoraproject.org//packages/plasma-desktop/6.1.3/2.fc40/x86_64/plasma-desktop-6.1.3-2.fc40.x86_64.rpm</a></td>
			</tr><tr>
				<th>Source</th>
				<td><a href="https://koji.fedoraproject.org/koji/rpminfo?rpmID=39405653">https://koji.fedoraproject.org/koji/rpminfo?rpmID=39405653</a></td>
			</tr></table>
		</td>
	</tr></table>
	
	</details>

1.	[x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting/b4ae37f563a62ecaa07d6a9a631b681cb82836ff#troubleshooting-blueman) and done my best effort to follow.

#### **Request**

As https://github.com/ProtonMail/proton-bridge/issues/475#issuecomment-2324867533 explains, it would function a lot better in custom colour schemes. That's why AOSP mandates monochromatic icons for its status bar and new Material You application icons.

https://github.com/blueman-project/blueman/issues?q=sort%3Aupdated-desc+is%3Aissue+is%3Aopen+monochrom returned 0 results, so I doubt that this is a duplicate.

cschramm, 2024-09-02:
You might be looking for `gsettings set org.blueman.general symbolic-status-icons true` to enable a symbolic status icon.

RokeJulianLockhart, 2024-09-02:
> https://github.com/blueman-project/blueman/issues/2479#issuecomment-2324888762

Indeed, I was, @cschramm. Thanks. Is that preference exposed via the GUI?

closed 2024-09-02 by RokeJulianLockhart

cschramm, 2024-09-02:
The primary purpose is distribution configuration, but user exposure would make sense, yes. I don't see why it should be enabled by default. Distributions that use symbolic icons can set the default themselves and otherwise fullcolor seems like a reasonable default. Monochromatic icon themes typically override the fullcolor icon accordingly.

cschramm, 2024-09-02:
> > [#2479 (comment)](https://github.com/blueman-project/blueman/issues/2479#issuecomment-2324912023)
> 
> @cschramm, I've not used a distribution that included `blueman` thus far, and on some OSes - like openSUSE Tumbleweed - the OS doesn't set these preferences, because it doesn't come bundled with a specific DE (it's chosen at installation time by the user).

Yes, if the user chooses the DE or icon theme, he's typically left with setting the option himself. That's where it makes sense to provide it in the UI.

> > Monochromatic icon themes typically override the fullcolor icon accordingly.
> 
> Can you elaborate on what you mean by this?

Papirus, for example, ships its own blueman icons; links to their monochrome bluetooth icons.