blueman-project/blueman #2484: Blueman manager 2.3.5-2+b1 not start after upgrade to Debian 12 (bookworm)

itmagpro, 2024-09-08:
blueman: 2.3.5-2+b1  i386
BlueZ: 5.66-1+deb12u2  i386
Distribution: Debian 12
Desktop environment: Xfce

Blueman manager 2.3.5-2+b1 not start after upgrade to Debian 12 (bookworm)

Output commands see here:

# dpkg -l|grep blue
$ blueman-adapters
$ blueman-manager

https://pastebin.com/uUJCJAAj

I try purge and reinstall bluez and blueman, but this not help ;(

dmesg about Bluetooth hci0 here: https://pastebin.com/JGfLpPxR

game009f, 2024-09-12:
我的是linuxmint 22，linux6.8.0.38，blueman 2.4
连接蓝牙后，修改蓝牙备注为中文，重启电脑后blueman就不能启用蓝牙
最后手动修改配置文件中的中文为英文才能正常启动蓝牙
/var/lib/bluetooth/xxxx/xxxx/info

cschramm, 2024-09-12:
`blueman-applet.service` is a user service, but just run `blueman-applet` (probably at `/usr/bin/blueman-applet`). It usually gets started automatically when requested due to `org.blueman.Applet.service` (`/usr/share/dbus-1/services/org.blueman.Applet.service`, I guess). There might be something wrong with your D-Bus setup.

itmagpro, 2024-09-21:
What "_something wrong with your D-Bus setup_", for example?

I have in system initscripts* insserv* sysv-rc* pkg installed, may be in your program some problem with sysv?

For example when reinstall bluez:

```
# apt-get install --reinstall bluez bluez-tools bluez-firmware bluez-obexd
...
Настраивается пакет bluez (5.66-1+deb12u2) …
insserv: Script `ssh' has overlapping Default-Start and Default-Stop runlevels (2 3 4 5) and (2 3 4 5). This should be fixed.
Обрабатываются триггеры для man-db (2.11.2-2) …
```

See:

- [#971713 - sysstat: init or systemd file has overlapping runlevels - Debian Bug report logs](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=971713)

But **apt-get install bluetooth blueman** no have this problem, also before reinstall I try:

```
# sctl stop dbus-org.bluez
# sctl disable dbus-org.bluez
# sudo update-rc.d -f dbus-org.bluez remove
# sudo update-rc.d dbus-org.bluez defaults

# sctl stop bluetooth
# sctl disable bluetooth
Synchronizing state of bluetooth.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable bluetooth
insserv: warning: current start runlevel(s) (empty) of script `bluetooth' overrides LSB defaults (2 3 4 5).
insserv: warning: current stop runlevel(s) (0 1 2 3 4 5 6) of script `bluetooth' overrides LSB defaults (0 1 6).
# sudo update-rc.d -f bluetooth remove
# sudo update-rc.d bluetooth defaults

# agi --reinstall bluetooth blueman bluez bluez-tools bluez-firmware bluez-obexd
...
without "insserv: warning"
#
```

Seems not in sysv or dbus problem because all other work fine, only blueman not start and bluetooth not work ;(

**P.S. I known, that NTO NEED BE UPGRADE TO DEBIAN 12 and need litle wait...**

- network-manager, wicd gui applet drop;
- bluetooth, blueman fail;
- grow consumption RAM;
- ..."GREAT" works...

cschramm, 2024-11-25:
I don't really get the problem statement. You provided some data that `blueman-manager` fails to start `blueman-applet` via D-Bus. It seems like you are able to run `blueman-applet` yourself. Does `blueman-manager` still complain about the applet service when it's already running?

itmagpro, 2025-01-11:
help self by my self :)

[blueman-manager, blueman-applet: Not loading conflicting plugin NMPANSupport (SOLVED)](https://itmag.pro/news/network/blueman-nmpansupport-conflict)

closed 2025-01-11 by itmagpro