blueman-project/blueman #2496: Bluetooth visibility

StevenGamer2006, 2024-09-21:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

thiagomandrik, 2024-09-21:
same here since yesterday

thiagomandrik, 2024-09-24:
I found this partial solution:

    sudo apt purge bluez
    sudo apt install bluez
    sudo apt install blueman
    Reboot

It works, but the bug reappears after the next boot.

Source:
https://forums.linuxmint.com/viewtopic.php?t=425923

infirit, 2024-09-27:
Looks like this was something in mint so closing.

ewinITservices, 2024-09-29:
I am currently using Linux Mint Cinnamon 22 and have been facing difficulties since yesterday. I tried connecting my Bluetooth speaker to both my Android phone and Windows 10, and it worked well on both devices. However, it does not seem to connect on Mint 22.

closed 2024-09-30 by infirit