blueman-project/blueman #2497: Blueman Tray is not using StatusNotifierItem correctly

ldelossa, 2024-09-21:
blueman: 2.4.3
BlueZ:
Distribution: Fedora
Desktop environment: Sway

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Hello,

I'm writing an integration which uses the StatusNotifier DBus protocol to display tray icons. 

I believe Blueman is not using this protocol correctly. 

Blueman seems to be registering its StatusNotifierItem via it's DBUS object name, and not the bus name. 

For example:

```
** (way-shell:67420): DEBUG: 14:37:29.902: status_notifier_service.c:on_handle_register_item() called: /org/blueman/sni
```
This makes it impossible to find the bus name. 

By contract this is how DropBox registers its StatusNotifierItem 

```
** (way-shell:71357): DEBUG: 14:55:53.290: status_notifier_service.c:on_handle_register_item() called: org.kde.StatusNotifierItem-7-1
```

Which is aligned to the spec

```
Register a StatusNotifierItem into the StatusNotifierWatcher, in the form of its full name on the session bus, for instance org.freedesktop.StatusNotifierItem-4077-1. A StatusNotifierItem instance must be registered to the watcher in order to be noticed from both the watcher and the StatusNotifierHost instances. If the registered StatusNotifierItem goes away from the session bus, the StatusNotifierWatcher should automatically notice it and remove it from the list of registered services.
```

Is this by design?

infirit, 2024-09-22:
I don't know a lot about the StatusNotifier stuff but I had a quick look at what Pithos does and replicated it in blueman, which works for me on plasma. Does this conform better with how you interpret the spec @ldelossa ?

@cschramm did pretty much all of the hard work and research on this so they hopefully can comment on the design part.

```patch
diff --git a/blueman/main/indicators/StatusNotifierItem.py b/blueman/main/indicators/StatusNotifierItem.py
index 3ab75ad59..a2387a661 100644
--- a/blueman/main/indicators/StatusNotifierItem.py
+++ b/blueman/main/indicators/StatusNotifierItem.py
@@ -101,7 +101,7 @@ class StatusNotifierItemService(DbusService):
     ItemIsMenu = False
 
     def __init__(self, tray: BluemanTray, icon_name: str) -> None:
-        super().__init__(None, "org.kde.StatusNotifierItem", "/org/blueman/sni", Gio.BusType.SESSION,
+        super().__init__(None, "org.kde.StatusNotifierItem", "/StatusNotifierItem", Gio.BusType.SESSION,
                          {"Category": "s", "Id": "s", "IconName": "s", "Status": "s", "Title": "s",
                           "ToolTip": "(sa(iiay)ss)", "Menu": "o", "ItemIsMenu": "b"})
         self.add_method("Activate", ("i", "i"), "", lambda x, y: tray.activate_status_icon())
@@ -151,7 +151,7 @@ class StatusNotifierItem(IndicatorInterface):
         try:
             Gio.bus_get_sync(Gio.BusType.SESSION).call_sync(
                 self._SNI_BUS_NAME, "/StatusNotifierWatcher", self._SNI_INTERFACE_NAME,
-                "RegisterStatusNotifierItem", GLib.Variant("(s)", ("/org/blueman/sni",)),
+                "RegisterStatusNotifierItem", GLib.Variant("(s)", ("org.blueman.Tray",)),
                 None, Gio.DBusCallFlags.NONE, -1)
             watcher_expected = True
         except GLib.Error:
```

cschramm, 2024-09-22:
I cannot recall all details from three years ago, but https://github.com/blueman-project/blueman/pull/1382#issuecomment-849353653 describes the situation rather good:

* libappindicator and Ayatana pass an object path and use the sender of the registration call as the service to talk to.
* KNotifications passes a service name and uses `/StatusNotifierItem` as the object path.
* The specification adopts KNotifications' service name argument but does not specify an object path.
* All implementations that I looked at support both the appindicator and the KNotifications variants. Otherwise they would have limited compatibility.

I do not understand how the specification should work without an agreed object path like `/StatusNotifierItem`. Also, that does not seem very reasonable, as it e.g. limits things to one StatusNotifierItem per service. The patch mainly means to replace the appindicator flavor with that of KNotifications. Yes, it's closer to the specification, as a service name gets passed instead of an object name, but the specification just does not work due to the lack of an object path.

As far as I can tell, things did not develop (and I do not expect them to), so applications are left with choosing the appindicator or the KNotifications flavor and hosts need to support both.

ldelossa, 2024-09-22:
Thank you @cschramm 

I think that explains the situation well. Now I understand there are multiple supported specs here. I dont feel strongly about keeping this issue open. Its a shame we have varying specs to deal with, but no issue here since one was picked and stuck with.

closed 2024-09-22 by ldelossa