blueman-project/blueman #2505: Blueman Manager frozen device list can't remove a device

kiers, 2024-10-03:
blueman:2.3.5
BlueZ:5.64-0ubuntu1.3
Distribution:Linux Mint 21.1 Vera
Desktop environment:Cinnamon

Trying to remove a device to reset it, and i find it's frozen in the list. Hitting the "—" minus bar doesn't work, right-click "remove" doesnt' work.  Any suggestion?

econo202, 2024-10-23:
- blueman: 2.2.4-1
- bluez: 5.64-0ubuntu1.3

I've the same problem that probably started after upgrading bluez from 5.64-0ubuntu1.1 to 5.64-0ubuntu1.3 (see [changelog in Ubuntu](https://changelogs.ubuntu.com/changelogs/pool/main/b/bluez/bluez_5.64-0ubuntu1.3/changelog)).

My workaround is to use `bluetoothctl`.

kiers, 2024-10-23:
Yeah...been using systemctl start/stop bt....
why should update mess with bluetooth so often?

cschramm, 2024-10-23:
Please run `blueman-manager --loglevel debug`, try to remove the device in the new window and check if anything shows in the output. The `bluetooth.service` logs might also be worth a quick look.

closed 2024-11-07 by github-actions[bot]

econo202, 2024-11-13:
I did little testing, and after downgrading bluez from `5.64-0ubuntu1.3` to `5.64-0ubuntu1` everything works normally.

kiers, 2024-11-13:
Even after downgrading bluez and bluez-obexd the same prevails.  It used to work flawlessly, then some update just kills it.
systemctl status bluetooth retuns:
 bluetoothd[5978]: Failed to set mode: Failed (0x03)

Trying to search bluetooth from blueman-manager produces:
File "/usr/lib/python3/dist-packages/blueman/bluez/Base.py", line 78 in callback...etc etc...GDBus.Error:org.bluez.Error.NotReady:REsource Not Ready (36).

econo202, 2024-11-14:
After more testing seems that setting `ClassicBondedOnly=false` in `/etc/bluetooth/input.conf` does the trick with bluez `5.64-0ubuntu1.3`, as described [in this bluez issue](https://github.com/bluez/bluez/issues/673).