blueman-project/blueman #2507: Authorisation in Hyprland

Ranger-Dj, 2024-10-10:
When i connect my headphones to pc via blueman , it connects and ask for authorisation I am using hyrplandit gives a notification about remote authorisation permission but it is read only , i cant allow or dismiss permission in hyprland

Jark5455, 2024-10-23:
I personally use mako and observed this same problem, I assumed it was just a mako issue and not hyprland specifically

cschramm, 2024-10-23:
From what I read mako does not render buttons for actions but they can be invoked by keybindings or `makoctl`. The go-to solution seems to be https://github.com/emersion/mako/issues/4#issuecomment-1723290536 (also with `on-button-left` and / or an additional `makoctl dismiss`, see https://github.com/emersion/mako/issues/283#issue-646646085)

It's a bit unfortunate that a notification daemon indicates actions support (which it obviously has) but does not show any indication of such actions by default...

closed 2024-10-23 by cschramm

Ranger-Dj, 2024-11-15:
it was with mako

Ranger-Dj, 2024-11-15:
but i changed setup it stopped getting issue