blueman-project/blueman #2510: Blueman is suddenly not working

vmxt, 2024-10-15:
blueman: 2.4.3-1
BlueZ: 5.78-1
Distribution: Arch Linux
Desktop environment: xfce4

* ✅ I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I went through the troubleshooting steps, but nothing worked. I'm using a bluetooth dongle because my board lacks built-in Bluetooth. It was working earlier, but now Bluetooth is not working. The blueman app cannot click search, despite the fact that I have started, enabled, and restarted the bluetooth services. I tried the manual `rfkill unblock blutooth` and `modprobe btusb`; it works perfectly on status and actively running, but there is no trace of working bluetooth.

bluetooth status
[image]

blueman
[image]

What went missing? Any hints?

cschramm, 2024-10-16:
What does `bluetoothctl list` show?

vmxt, 2024-10-16:
Hi @cschramm , as requested to answer your ff. questions

> What does `bluetoothctl list` show?

Nothing shows

> Also please provide the output of `blueman-applet --loglevel debug` with blueman-applet initially not running.

```
blueman-applet 15.15.32 INFO     PluginManager:86 load_plugin: ['PPPSupport', 'DiscvManager', 'AuthAgent', 'DhcpClient', 'StatusNotifierItem', 'StandardItems', 'ExitItem', 'AutoConnect', 'GameControllerWakelock', 'DisconnectItems', 'PulseAudioProfile', 'DBusService', 'StatusIcon', 'Menu', 'SerialManager', 'TransferService', 'ConnectionNotifier', 'NMDUNSupport', 'Networking', 'KillSwitch', 'RecentConns', 'NetUsage', 'NMPANSupport', 'PowerManager', 'ShowConnected']
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.PulseAudioProfile.AudioProfiles'>
blueman-applet 15.15.32 INFO     PulseAudioUtils:207 pa_context_event: 1
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 15.15.32 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 15.15.32 INFO     PulseAudioUtils:207 pa_context_event: 2
blueman-applet 15.15.32 INFO     KillSwitch:106 io_event  : killswitch registered 0
blueman-applet 15.15.32 INFO     KillSwitch:122 io_event  : State: False
blueman-applet 15.15.32 INFO     PowerManager:161 update_power_state: off True | foff False | on True | current state True | new state False
blueman-applet 15.15.32 INFO     PowerManager:164 update_power_state: Signalling False
blueman-applet 15.15.32 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 15.15.32 INFO     TransferService:243 _on_dbus_name_vanished: org.bluez.obex not running or was stopped
blueman-applet 15.15.32 INFO     Applet:76 _on_dbus_name_appeared: org.bluez :1.6
blueman-applet 15.15.32 INFO     Functions:99 launch    : Gtk eventtime is 0, not using LaunchContext
blueman-applet 15.15.32 WARNING  DiscvManager:122 update_menuitems: warning: Adapter is None
blueman-applet 15.15.32 INFO     BluezAgent:56 register_agent: Register Agent
blueman-applet 15.15.32 INFO     Networking:78 set_nap   : set nap False
blueman-applet 15.15.32 INFO     ShowConnected:55 enumerate_connections: Found 0 existing connections
blueman-applet 15.15.32 INFO     PulseAudioUtils:207 pa_context_event: 3
blueman-applet 15.15.32 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.ShowConnected.ShowConnected object at 0x7bde898a7920>
blueman-applet 15.15.32 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.NMPANSupport.NMPANSupport object at 0x7bde849ff860>
blueman-applet 15.15.32 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.KillSwitch.KillSwitch object at 0x7bde849ffa10>
[... cut ...]
blueman-applet 15.15.32 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <StatusIcon.StatusIcon object at 0x7bde84a23d80 (blueman+plugins+applet+StatusIcon+StatusIcon at 0x5b5efb2614d0)>
blueman-applet 15.15.32 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.StandardItems.StandardItems object at 0x7bde849fef60>
blueman-applet 15.15.32 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.Menu.Menu object at 0x7bde8bb938c0>
blueman-applet 15.15.32 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DBusService.DBusService object at 0x7bde898d3470>
```

cschramm, 2024-10-16:
Your system does not provide any adapter then. Probably some issue following a kernel or firmware update. Check `dmesg` for Bluetooth issues.

I'm surprised to see the power toggle enabled in the manger window under that condition. :thinking: CC @infirit

vmxt, 2024-10-16:
running `dmesg | grep -i bluetooth` and displays

```
[    3.807380] Bluetooth: Core ver 2.22
[    3.807407] NET: Registered PF_BLUETOOTH protocol family
[    3.807409] Bluetooth: HCI device and connection manager initialized
[    3.807412] Bluetooth: HCI socket layer initialized
[    3.807415] Bluetooth: L2CAP socket layer initialized
[    3.807419] Bluetooth: SCO socket layer initialized
[    3.839787] Bluetooth: hci0: CSR: Setting up dongle with HCI ver=9 rev=0810
[    3.839792] Bluetooth: hci0: LMP ver=9 subver=2312; manufacturer=10
[    3.839793] Bluetooth: hci0: CSR: Unbranded CSR clone detected; adding workarounds and force-suspending once...
[    4.171680] Bluetooth: hci0: HCI Delete Stored Link Key command is advertised, but not supported.
[    4.171684] Bluetooth: hci0: HCI Read Default Erroneous Data Reporting command is advertised, but not supported.
[    4.171686] Bluetooth: hci0: HCI Set Event Filter command not supported.
[    6.177389] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    6.177393] Bluetooth: BNEP filters: protocol multicast
[    6.177397] Bluetooth: BNEP socket layer initialized
[    6.210251] Bluetooth: hci0: Opcode 0x1004 failed: -110
[    6.210254] Bluetooth: hci0: command 0x1004 tx timeout
```

infirit, 2024-10-16:
> I'm surprised to see the power toggle enabled in the manger window under that condition. 🤔 CC @infirit

As am I. I'll have a look.

Shubham-550, 2024-10-16:
@vmxt I was having the same issue on (Arch + Hyprland). I looked in the Linux forums but could not find any solution. Then I googled for Bluetooth problems related to my laptop, and on Reddit I found a workaround/solution.
`60 second power button hold down` works for me.

https://www.reddit.com/r/LenovoLegion/comments/qr4mp6/sudden_onset_bluetooth_issues_legion_5_pro/

vmxt, 2024-10-16:
> @vmxt the kernel fails to initialize the adapter (the timeout). If you updated your kernel recently try reverting to the previous and see if it works there. Your adapter is based on what usually is called a "fake" CSR chip. These tend to break and get fixed in the kernel.

Hi @infirit I reverted back to : Linux archlinux 6.11.3-arch1-1 kernel. I did the `restart`, `unblock`, and `debug` on troubleshoot still no good

and still getting the same logs
```
[    5.630835] Bluetooth: Core ver 2.22
[    5.630853] NET: Registered PF_BLUETOOTH protocol family
[    5.630854] Bluetooth: HCI device and connection manager initialized
[    5.630858] Bluetooth: HCI socket layer initialized
[    5.630861] Bluetooth: L2CAP socket layer initialized
[    5.630864] Bluetooth: SCO socket layer initialized
[    5.817216] Bluetooth: hci0: CSR: Setting up dongle with HCI ver=9 rev=0810
[    5.817219] Bluetooth: hci0: LMP ver=9 subver=2312; manufacturer=10
[    5.817220] Bluetooth: hci0: CSR: Unbranded CSR clone detected; adding workarounds and force-suspending once...
[    6.178281] Bluetooth: hci0: HCI Delete Stored Link Key command is advertised, but not supported.
[    6.178285] Bluetooth: hci0: HCI Read Default Erroneous Data Reporting command is advertised, but not supported.
[    6.178286] Bluetooth: hci0: HCI Set Event Filter command not supported.
[    6.609816] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    6.609820] Bluetooth: BNEP filters: protocol multicast
[    6.609824] Bluetooth: BNEP socket layer initialized
[    8.360267] Bluetooth: hci0: Opcode 0x1004 failed: -110
```
blueman system tray appearing but disappear when enabled
[image]

vmxt, 2024-10-16:
> @vmxt I was having the same issue on (Arch + Hyprland). I looked in the Linux forums but could not find any solution. Then I googled for Bluetooth problems related to my laptop, and on Reddit I [...]

hey @Shubham-550 , did this yesterday but still no good.

Tom-jm03, 2024-10-16:
Doing a full power shutdown (Hold power button for 60 seconds, then turning power switch from power supply off and pressing the power button a couple of times to remove all left over electricity, switching it back on) fixed the issue for me.
`Device 008: ID 8087:0029 Intel Corp. AX200 Bluetooth`

vmxt, 2024-10-16:
> Doing a full power shutdown (Hold power button for 60 seconds, then turning power switch from power supply off and pressing the power button a couple of times to remove all left over electricity, switching it back on) fixed the issue for me.
> 
> Device: Device 008: ID 8087:0029 Intel Corp. AX200 Bluetooth
> 
> Distro: Arch Linux x86_64 6.11.3-arch1-1

Hi @Tom-jm03, did this on the 4th time. and still not working. haha

vmxt, 2024-10-16:
**ADDITIONAL** yea on this command looks like the bluetooth dongle was recognized. but no worries. I'm still looking for this solution.

```
[jefferson@archlinux ~]$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 004: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
Bus 001 Device 005: ID 3151:4010 ROYUAN MonsGeek Keyboard
Bus 001 Device 006: ID 0458:6006 KYE Systems Corp. (Mouse Systems) USB 2.0 HD1080P PC Camera
Bus 001 Device 007: ID 0458:0186 KYE Systems Corp. (Mouse Systems) Genius DX-120 Mouse
Bus 001 Device 010: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

rakibdev, 2024-10-17:
Downgrading Linux from 6.11.3 to 6.11.1 fixed it.
https://bugzilla.kernel.org/show_bug.cgi?id=219365

vmxt, 2024-10-17:
> Downgrading Linux from 6.11.3 to 6.11.1 fixed it. https://bugzilla.kernel.org/show_bug.cgi?id=219365

It didn't work in kernels `6.11.3` and `6.11.2`! kudos to you @rakibdev works perfectly fine on version `6.11.1`. I didn't test this version because I assumed it would be the same result.

closed 2024-10-17 by vmxt

nicoestrada, 2024-10-20:
I downgraded from 6.11.4 to 6.11.3 but downgrading kernels seems to be the solution

vmxt, 2024-10-20:
> I downgraded from 6.11.4 to 6.11.3 but downgrading kernels seems to be the solution

yea that is the only solution at the moment. so annoying to see ignoring packages when updating. haha