blueman-project/blueman #2521: blueman-applet excessive CPU usage after resume

tommie, 2024-10-23:
blueman: 2.3.5-3build1 (applet c07e8a467f2988a028da80b2d9ff135a8371d093)
BlueZ: 5.72-0ubuntu5
Distribution: Ubuntu 24.04
Desktop environment: Gnome

This is a continuation on the discussion in #2499.

Today blueman-applet went into using constant excessive CPU after suspend/resume. It's at 10% rather than 100%, but that might be due to debug logging slowing it down.

https://gist.github.com/tommie/0ec3fa00f013995ab87f6e136d56fac7

It seems to be flooding

    blueman-applet 07.50.43 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_04_B9_E3_48_CF_87 {'RSSI': -85}

with different RSSI for the same device. Normally this only happens 1-2 seconds per minute as devices are added/removed, but now it keeps running constantly, logging several times per second. It keeps happening even if I restart blueman-applet.

The assertion error

```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/bluez/Manager.py", line 76, in _on_object_removed
    assert isinstance(device_proxy, Gio.DBusProxy)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AssertionError
```

seems unrelated as it occurs previously in the log file without causing the behavior.

infirit, 2024-10-23:
We get these RSSI property changes from BlueZ's dbus api for devices when discovery is in progress. And I've never seen so many in such a short time ever. Like you said, normally BlueZ would update every couple seconds.

Is there anything interesting in the kernel log `journalctl --dmesg`? Or maybe in the bluetoothd daemon log `journalctl -u bluetooth.service`.

The assertion error should not happen but afaics harmless.

tommie, 2024-10-23:
Kernel logs are very quiet about Bluetooth. bluez has nothing that seems related. It complains about my Bose headphones as I turn them on and off (and sometimes picks the wrong audio profile.)

I killed the applet and restarted it a couple of times. Right now it's off, and now `/usr/lib/x86_64-linux-gnu/indicator-bluetooth/indicator-bluetooth-service` is at 82% CPU and bluetoothd is around 6%. If I kill indicator-bluetooth, bluetoothd still sits around 6%.

So perhaps this is rooted in bluez having a mental breakdown.

tommie, 2024-10-23:
Just realized: I am running `bluetoothd --experimental` because I needed access to some experimental API at some point.

tommie, 2024-12-24:
(I've had blueman-applet disabled since my last report.)

closed 2025-03-02 by github-actions[bot]