blueman-project/blueman #2531: Allow window width of 512 or less for tiling support on 1024 wide screens

anohren, 2024-11-07:
blueman: 2.4.3
BlueZ:
Distribution: Debian testing (Trixie)
Desktop environment: MATE desktop on X11

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Im unable to drag the main blueman devices window to the screen edge and get it to resize to half the screen with a screen width of 1024. UI wise there seems to be plenty of room to spare but the window seems to refuse to go below ~520. I suggest allowing for a smaller window to enable tiling in more scenarios. At least my Bluetooth devices really don't have that long names 🙂

anohren, 2024-11-08:
I'm unsure if I understand what you'd like to know — the fact that I'm not seeing something different on MATE is the reason I'm here; you didn't explicitly mention the minimum width, but iOS Photos says that your image is 531 pixels wide, which is more than 512.

infirit, 2024-11-08:
If you see the same then there nothing we can do about it. As I said before we need to reserve space for all elements even if they are hidden for you.

anohren, 2024-11-08:
Ok, thanks. I just assumed it wouldn't be a problem since e.g. Caja suggested that collapsing items horizontally etc. was built into the toolkit, but I heard you loud and clear.

closed 2024-11-08 by anohren