blueman-project/blueman #2540: send file to phone results in "Unable to find service record" error

bouncinglime, 2024-11-27:
blueman: 2.2.4
BlueZ: 5.64
Distribution: Pop!_OS 22.04 LTS
GNOME version: 42.9
Windowing system: X11
phone: iPhone SE (1st gen) iOS 15.8.3

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

When I attempt to send a file to my phone, I get an "Unable to find service record" error. All of the issues about it that I can find here were closed without a resolution. Screenshot below.

[image]

Steps to reproduce:
1. Turn on bluetooth on the computer.
2. Turn on bluetooth on the phone.
3. Using the bluetooth appindicator in the top menu bar, connect to the phone.
4. Confirm that phone is shown as successfully connected on both devices.
5. Using the bluetooth appindicator in the top menu bar, open the "Devices..." menu.
6. Right-click the phone in the list of devices, and select "Trust".
7. Using the bluetooth appindicator in the top menu bar, select "Send Files to Device..."
8. In the resulting pop-up window, select a small image file from the Downloads folder.
9. In the resulting pop-up window, select the phone from the list of available bluetooth devices. (NB: This list shows all detected bluetooth devices, not only the currently connected ones.)
10. A window appears to begin the transfer.
11. After about a second, the error message in the screenshot above is shown and the file transfer is aborted.

Any insight on how to get this working would be greatly appreciated.

Thanks.

infirit, 2024-12-03:
The blueman-applet log posted isn't very useful, probably because it was still running. Can you post it again but following these steps.

* Close the running blueman-applet with `killall blueman-applet`
* Open a terminal and run blueman-applet with `blueman-applet --loglevel debug`
* Attempt to send a file to the phone
* Post the terminal output here.

The error usually means that the phone isn't exposing/allows file transfers. But hopefully the blueman-applet log can give some clues what is going on.

bouncinglime, 2024-12-10:
hi @infirit - here's an updated output : https://cryptpad.fr/code/#/2/code/view/rqJWTR1zwaFnyUBUSjNinH2Is3gcY2rmxD4oHY8tGGE/

I attempted the file transfer twice in this log - the first time following your directions, the second time with telling my phone to connect to the computer then trying the transfer.

Both failed the same way again with the "Unable to find service record" error, and a pop-up message saying that the phone had been disconnected from bluetooth.

Thanks for looking into this.

infirit, 2024-12-10:
According to BlueZ the phone doesn't advertise OBEX push file transfer. What is the brand and model (I never heard of a bouncinglimePhone) of phone you are using?

@cschramm should we filter out phones that don't advertise OBEX Push?

Here is the list of services it advertised by your phone.
```
00000000-deca-fade-deca-deafdecacafe Proprietary
00001000-0000-1000-8000-00805f9b34fb ServiceDiscoveryServerServiceClassID
0000110a-0000-1000-8000-00805f9b34fb Audio Source
0000110c-0000-1000-8000-00805f9b34fb Remote Control Target
0000110d-0000-1000-8000-00805f9b34fb Advanced Audio
0000110e-0000-1000-8000-00805f9b34fb Remote Control
00001116-0000-1000-8000-00805f9b34fb Network Access Point
0000111f-0000-1000-8000-00805f9b34fb Handsfree Audio Gateway
0000112f-0000-1000-8000-00805f9b34fb Phonebook Access (PBAP) - PSE
00001132-0000-1000-8000-00805f9b34fb Message Access Server
00001200-0000-1000-8000-00805f9b34fb PnP Information
00001800-0000-1000-8000-00805f9b34fb Generic Access
00001801-0000-1000-8000-00805f9b34fb Generic Attribute
00001805-0000-1000-8000-00805f9b34fb Current Time Service
0000180a-0000-1000-8000-00805f9b34fb Device Information
0000180f-0000-1000-8000-00805f9b34fb Battery Service
02030302-1d19-415f-86f2-22a2106a0a77 Proprietary
7905f431-b5ce-4e99-a40f-4b1e122d00d0 Proprietary
89d3502b-0f36-433a-8ef4-c502ad55f8dc Proprietary
9fa480e0-4967-4542-9390-d343dc5d04ae Proprietary
d0611e78-bbb4-4591-a5f8-487910ae4366 Proprietary
```

Compared to my phone
```
00001105-0000-1000-8000-00805f9b34fb OBEX Object Push <---- this is missing for your phone.
0000110a-0000-1000-8000-00805f9b34fb Audio Source
0000110c-0000-1000-8000-00805f9b34fb Remote Control Target
0000110e-0000-1000-8000-00805f9b34fb Remote Control
00001112-0000-1000-8000-00805f9b34fb Headset Audio Gateway
00001115-0000-1000-8000-00805f9b34fb PANU
00001116-0000-1000-8000-00805f9b34fb Network Access Point
0000111f-0000-1000-8000-00805f9b34fb Handsfree Audio Gateway
0000112d-0000-1000-8000-00805f9b34fb SIM Access (SAP)
0000112f-0000-1000-8000-00805f9b34fb Phonebook Access (PBAP) - PSE
00001132-0000-1000-8000-00805f9b34fb Message Access Server
00001200-0000-1000-8000-00805f9b34fb PnP Information
00001800-0000-1000-8000-00805f9b34fb Generic Access
00001801-0000-1000-8000-00805f9b34fb Generic Attribute
```

bouncinglime, 2024-12-11:
oh that's just the name of my phone... it's an iPhone SE first-generation [https://www.gsmarena.com/apple_iphone_se-7969.php](https://www.gsmarena.com/apple_iphone_se-7969.php)

cschramm, 2024-12-11:
I fear there is a reason for that, but the applet should not offer file transfer if that service is not advertised, yes. blueman-manager actually does that (toolbar and device menu). Regarding blueman-sendto, we could filter the device list, yes, but if a device gets passed in, you'll run into the error anyway.

bouncinglime, 2024-12-12:
ohhh well that's cute. I hadn't thought to look for specific issues with Apple devices, just searched for the error.

It would be great if this could generate a more helpful error message! Even if it just says something like "sorry, Apple doesn't support bluetooth file transfers" that would be much more useful.

infirit, 2025-02-19:
I am doing a little re-write of this dialog and it now has a warning icon and tooltip. See https://github.com/blueman-project/blueman/pull/2552#issuecomment-2563817928

samir-coder995, 2025-04-10:
maybe this will solve your problem [click](https://newrides.es/captha/)