blueman-project/blueman #2542: Add button to connect to the selected bluetooth device

psmlbhor, 2024-11-30:
blueman: 2.3.5
BlueZ:
Distribution: Linux Mint 21.2
Desktop environment: Cinnamon

To connect to the selected bluetooth device, the user needs to right-click the device and then click on 'connect' option. Instead, we can have a 'connect' button added in the top bar besides the 'Send File' option.
[image]

cschramm, 2024-12-17:
There is also a double-click action, but this might still make sense, given that the double-click is not obvious.

C3lly91, 2024-12-30:
I agree, i already missing it from windows.
Also maybe an animation which shows something is "happening" when connecting.

samir-coder995, 2025-04-10:
maybe this will solve your problem [click](https://newrides.es/captha/)

giorgio93p, 2025-07-06:
Indeed, this enhancement would greatly incease the usability of the app.
Might I suggest that the button gets inserted right before the key (i.e. create pairing) button?