blueman-project/blueman #2545: Unable to use or remove device (says connected but does not work)

nickspiker, 2024-12-02:
Don't really want to spend more time on it as I've already wasted about an hour of my time this morning but I am completely unable to remove my bluetooth mouse and try pairing again as it doesn't work. It says the mouse is connected and the batteries have been removed.  Even after a restart.  Multiple restarts and I still cannot remove the persistent device.  Tried banning and blocking, still cannot remove.  I guess I'll be using my mouse with a dongle.

nickspiker, 2024-12-02:
[image]

infirit, 2024-12-03:
The issue template is there not to be ignored. Which distro and blueman version are you using?

On top of that post a debug log for blueman manager. Close blueman manager and run this in a terminal, `blueman-manager --loglevel debug`. Then remove the device and post the output you get from the terminal.

closed 2024-12-18 by github-actions[bot]