blueman-project/blueman #2551: Applet non working with Glib2 2.82.3 on slackel (slackware based)

soyDESKTOP68A2MRFT, 2024-12-10:
blueman: 2.4.3
BlueZ: 5.79
Distribution: Slackel
Desktop environment: LXDE

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I recently updated glib2 to 2.82.3, and the applet isn't working since then, here is a paste.rs paste, with the logs:

https://paste.rs/yesnV.bash

lukeemhigh, 2024-12-10:
I think I can reproduce on Arch Linux. I recently updated glib2 to 2.4.3 and `blueman-applet` started failing.

blueman 2.4.3
bluez 5.79

Here's the debug log obtained from running `blueman-applet --loglevel debug`:

```sh
blueman-applet 23.40.53 INFO     PluginManager:86 load_plugin: ['DiscvManager', 'GameControllerWakelock', 'PulseAudioProfile', 'DisconnectItems', 'SerialManager', 'AuthAgent', 'StatusNotifierItem', 'StatusIcon', 'PPPSupport', 'DBusService', 'ConnectionNotifier', 'TransferService', 'DhcpClient', 'ExitItem', 'RecentConns', 'Networking', 'Menu', 'PowerManager', 'AutoConnect', 'ShowConnected', 'StandardItems', 'KillSwitch', 'NMPANSupport', 'NetUsage', 'NMDUNSupport']
blueman-applet 23.40.53 WARNING  PluginManager:93 load_plugin: Failed to start plugin GameControllerWakelock: Only X11 platform is supported
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.PulseAudioProfile.AudioProfiles'>
blueman-applet 23.40.53 INFO     PulseAudioUtils:207 pa_context_event: 1
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 23.40.53 WARNING  PluginManager:150 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 23.40.53 WARNING  PluginManager:150 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 23.40.53 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 23.40.53 INFO     PulseAudioUtils:207 pa_context_event: 2
blueman-applet 23.40.53 INFO     KillSwitch:106 io_event  : killswitch registered 0
blueman-applet 23.40.53 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 23.40.53 INFO     PowerManager:161 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 23.40.53 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 23.40.53 INFO     TransferService:243 _on_dbus_name_vanished: org.bluez.obex not running or was stopped
blueman-applet 23.40.53 INFO     Applet:76 _on_dbus_name_appeared: org.bluez :1.3
blueman-applet 23.40.53 INFO     Functions:99 launch    : Gtk eventtime is 0, not using LaunchContext
blueman-applet 23.40.53 INFO     BluezAgent:56 register_agent: Register Agent
blueman-applet 23.40.53 INFO     Networking:78 set_nap   : set nap False
blueman-applet 23.40.53 INFO     PulseAudioUtils:207 pa_context_event: 3
blueman-applet 23.40.53 DEBUG    Base:67 _properties_changed: /org/bluez/hci0 {'Pairable': True} [] <Adapter.Adapter object at 0x75fb3ac688c0 (blueman+bluez+Adapter+Adapter at 0x5d84295d26b0)>
blueman-applet 23.40.53 DEBUG    DiscvManager:95 on_adapter_property_changed: prop Pairable True
blueman-applet 23.40.53 INFO     ShowConnected:55 enumerate_connections: Found 0 existing connections
blueman-applet 23.40.53 INFO     PulseAudioUtils:207 pa_context_event: 4
blueman-applet 23.40.53 INFO     PulseAudioProfile:135 on_pa_ready: PulseAudio Ready
[... cut ...]
  File "/usr/lib/python3.12/site-packages/blueman/main/Tray.py", line 42, in _on_name_appeared
    self.indicator = indicator_class(self, applet.GetIconName())
                                           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/gi/overrides/Gio.py", line 374, in __call__
    result = self.dbus_proxy.call_sync(self.method_name, arg_variant,
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
gi.repository.GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Remote peer disconnected (4)
```

smtucker, 2024-12-10:
I am also having the same issue on Arch with blueman 2.4.3 and bluez 5.79. Downgrading from glib2-2.82.3 to glib2-2.82.2 resolves the issue.

cschramm, 2024-12-11:
I am pretty sure you're experiencing https://gitlab.gnome.org/GNOME/glib/-/issues/3559. There will be a 2.82.4 release probably today.

closed 2024-12-11 by cschramm