blueman-project/blueman #2559: Connection Failed: Unknown error

mattgu61, 2024-12-17:
Distribution: lime
Desktop environment: 

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I have this error message Connection Failed: Unknown error when I tried to connect my headset in bluetooth. It connects but then I receive the error message and disconnect my device. I have a workaround of unpairing my headset and pairing it again to work but it's not ideal. This happened after the last update I did. Thank you.
-->

crhy, 2024-12-17:
Yes I'm also having this problem quite frequently.  It's definitely mint.  the BT device in question hooks up pretty much immediately to all other devices.

Bierchermuesli, 2024-12-20:
Same here. Intel Corp. AX201

```
blueman-applet --loglevel debug
blueman-applet 10.17.31 INFO     RecentConns:112 notify    : <Device.Device object at 0x79b7f454ee40 (blueman+bluez+Device+Device at 0x8a729b0)> 00000000-0000-0000-0000-000000000000
blueman-applet 10.17.34 DEBUG    Base:67 _properties_changed: /org/bluez/hci0/dev_AC_BF_71_CC_9D_8B {'Connected': True} [] <Device.Device object at 0x79b7f454ee40 (blueman+bluez+Device+Device at 0x8a729b0)>
blueman-applet 10.17.39 DEBUG    Base:67 _properties_changed: /org/bluez/hci0/dev_AC_BF_71_CC_9D_8B {'Connected': False} [] <Device.Device object at 0x79b7f454ee40 (blueman+bluez+Device+Device at 0x8a729b0)>
```

Linux mint 6.8.0-51-generic
Intel AX201
Bose QuietComfort Ultra

ClarinetInALeatherJacket, 2024-12-23:
I have the same issue, using a Sennheiser Accentum. 

`blueman-applet 17.15.07 INFO     PluginManager:86 load_plugin: ['ConnectionNotifier', 'NMDUNSupport', 'Menu', 'DiscvManager', 'Networking', 'GameControllerWakelock', 'StatusNotifierItem', 'StatusIcon', 'NMPANSupport', 'TransferService', 'RecentConns', 'NetUsage', 'AutoConnect', 'KillSwitch', 'ShowConnected', 'SerialManager', 'ExitItem', 'StandardItems', 'DBusService', 'PulseAudioProfile', 'PowerManager', 'DisconnectItems', 'AuthAgent', 'DhcpClient', 'PPPSupport']
blueman-applet 17.15.07 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 17.15.08 WARNING  PluginManager:150 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.PulseAudioProfile.AudioProfiles'>
blueman-applet 17.15.08 INFO     PulseAudioUtils:207 pa_context_event: 1
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 17.15.08 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 17.15.08 WARNING  PluginManager:150 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 17.15.08 INFO     KillSwitch:106 io_event  : killswitch registered 0
blueman-applet 17.15.08 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 17.15.08 INFO     PowerManager:161 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 17.15.08 INFO     PulseAudioUtils:207 pa_context_event: 2
blueman-applet 17.15.08 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 17.15.08 INFO     TransferService:233 _on_dbus_name_appeared: org.bluez.obex :1.77
blueman-applet 17.15.08 INFO     Applet:76 _on_dbus_name_appeared: org.bluez :1.4
blueman-applet 17.15.08 INFO     Functions:99 launch    : Gtk eventtime is 0, not using LaunchContext
blueman-applet 17.15.08 INFO     Networking:78 set_nap   : set nap False
blueman-applet 17.15.08 INFO     BluezAgent:56 register_agent: Register Agent
blueman-applet 17.15.08 INFO     PulseAudioUtils:207 pa_context_event: 3
blueman-applet 17.15.08 INFO     AgentManager:17 on_registered: /org/bluez/obex/agent/blueman
blueman-applet 17.15.08 INFO     ShowConnected:55 enumerate_connections: Found 0 existing connections
blueman-applet 17.15.08 DEBUG    Base:67 _properties_changed: /org/bluez/hci0 {'Pairable': True} [] <Adapter.Adapter object at 0x74545cf2b900 (blueman+bluez+Adapter+Adapter at 0x2e072040)>
blueman-applet 17.15.08 DEBUG    DiscvManager:95 on_adapter_property_changed: prop Pairable True
blueman-applet 17.15.08 INFO     PulseAudioUtils:207 pa_context_event: 4
blueman-applet 17.15.08 INFO     PulseAudioProfile:134 on_pa_ready: PulseAudio Ready
blueman-applet 17.15.08 INFO     PulseAudioUtils:213 <lambda>  : 1
[... cut ...]
    2: deb http: //security.ubuntu.com/ubuntu jammy-security main restricted universe multiverse
  Active apt repos in: /etc/apt/sources.list.d/winehq-noble.sources
    1: deb [arch=amd64 i386] https: //dl.winehq.org/wine-builds/ubuntu noble main
Info:
  Memory: total: 32 GiB note: est. available: 31.09 GiB used: 6.28 GiB (20.2%)
  Processes: 459 Power: uptime: 15h 37m states: freeze,mem,disk suspend: s2idle wakeups: 1
    hibernate: platform Init: systemd v: 255 target: graphical (5) default: graphical
  Compilers: gcc: 13.3.0 Client: Cinnamon v: 6.2.9 inxi: 3.3.34
`

cschramm, 2024-12-23:
Audio devices are managed by your sound server, PipeWire in Linux Mint 22.