blueman-project/blueman #2570: Just bought and then installed Logictech MX Ergo S Mouse (trackball)

Bazmundi, 2024-12-18:
blueman:
BlueZ: [bluetoothctl --version](bluetoothctl: 5.64)
Distribution:
Desktop environment:

Processor	AMD Ryzen 9 5950X 16-Core Processor
Memory	131834MB (4289MB used)
Machine Type	Desktop
Operating System	Ubuntu 22.04.5 LTS
Date/Time	Wed 18 Dec 2024 23:08:34
Display
Resolution	1920x1080 pixels
OpenGL Renderer	NVIDIA GeForce RTX 4080/PCIe/SSE2
X11 Vendor	The X.Org Foundation
Audio Devices
Audio Adapter	HDA-Intel - HDA NVidia
Audio Adapter	HDA-Intel - HD-Audio Generic
Input Devices
Power Button	
Power Button	
Logitech M510	
Logitech K350	
HDA NVidia HDMI/DP,pcm:3	
HDA NVidia HDMI/DP,pcm:7	
HDA NVidia HDMI/DP,pcm:8	
HDA NVidia HDMI/DP,pcm:9	
HD-Audio Generic Front Mic	
HD-Audio Generic Rear Mic	
HD-Audio Generic Line	
HD-Audio Generic Line Out Front	
HD-Audio Generic Line Out Surround	
HD-Audio Generic Line Out CLFE	
HD-Audio Generic Front Headphone	
**_MX Ergo S Mouse_**	
Philips SHB5500 (AVRCP)	

Operating System
Version
Kernel	Linux 6.8.0-49-generic (x86_64)
Version	#49~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Nov 6 17:42:15 UTC 2
C Library	GNU C Library / (Ubuntu GLIBC 2.35-0ubuntu3.8) 2.35
Distribution	Ubuntu 22.04.5 LTS

* [* ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

https://gist.github.com/Bazmundi/bf77a7180b3c15040cad9eb421229405

Trackball is connected. Works fine.  Since connection, I am getting no blue-manager on indicator area.  I can open it from the Applications menu.  I am also getting sporadic exception errors and hitting the report error button.  When I check the app causing the problem is blueman.

infirit, 2024-12-29:
Make sure you did `killall blueman-applet` before running it from the terminal. You log suggest blueman-applet was still running.

Bazmundi, 2024-12-29:
blueman-manager was mentioned, not blueman-applet. When installed originally, blue-manager had a glyph in top right of screen so I could select that glyph to open the blueman-manager.  

In any event, on your suggestion, I used 'killall blueman-applet', then tried rebooting my machine to see if the icon for blueman-manager popped up againt top right.  It did not.

cschramm, 2024-12-30:
Your desktop environment would be vital to discuss issues with the tray icon. The things you dumped show that you're using Ubuntu whose default desktop would be Gnome. Gnome dropped tray icons in 2017. They seem to have been bottom left before that (by default). I'm not sure what you're referring to with "top right", "indicator area", or "glyph".