blueman-project/blueman #2572: Blueman won't search for any devices

atomi19, 2024-12-19:
blueman: 2.4
BlueZ:
Distribution: Linux Mint
Desktop environment: Cinnamon

Hi, i got some problem with blueman-manager, so it doesn't want search for any devices, only after reboot system, then maybe it will work, but not always. All buttons are not available(except exit)

[image]

romychs, 2024-12-22:
Same issue with Blueman: v2.3.5 and Linux Mint 21.3 Virginia, Cinnamon
USB-Bluetooth adapter
ID 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio
Under windows, device work normal.

yck24, 2024-12-23:
Gleiches Problemm mit Mint 22.
Ausserdem funktioniert auch LAN Adapter auch nicht. Realtek 2.5 gbps.

cschramm, 2024-12-23:
Anything in `bluetoothctl list`? Otherwise, your system does not have any adapter set-up.