blueman-project/blueman #2574: File transfer: existing files are silently overwritten, file renaming failure

Vrihub, 2024-12-20:
blueman: 2.3.5
BlueZ: 5.66
Distribution: Debian 12.8
Desktop environment: none

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Steps to reproduce:
- Transfer a file from an external device
- Repeat the previous step with the same file (or with another file having the same file name)
- In the first notification popup message (`"Incoming file over Bluetooth"`) the original file name is displayed, e.g. `bluetooth_content_share.html`
- In the final popup message (`"File Received"`) the file name has a timestamp string prepended, e.g. `20241220140923_bluetooth_content_share.html`
- The blueman log says: `TransferService:294 _on_transfer_completed: Destination file exists, renaming to: 20241220140923_bluetooth_content_share.html`

Expected result: the existing file should be left alone and the new copy should be saved under the modified file name (timestamp prepended)

Actual result: the existing file is silently overwritten.
It looks like the safety renaming mentioned in the log doesn't take place actually

Vrihub, 2024-12-20:
Thanks, I applied the patch in #2575 to my blueman installation and I can confirm that the bug is fixed: the old file is preserved and the new file is written using the correct filename.

I also get a couple of small side effects, though:
- I get a new popup notification message: `"Files received: Received -1 files in the background"`. I've never seen this message before
- The final popup notification message says `"Transfer failed: Transfer of file 20241220202409_bluetooth_content_share.html failed"`, even if the transfer was actually successful (see above)
- (I also get a FileNotFoundError exception and other related errors in blueman's log)

I presume this is because I'm applying the patch to an old 2.3.5 installed release? Or is it because more work is needed instead? Let me know if I can help you (though I can't install-test the current HEAD on my system, sorry)

infirit, 2024-12-20:
Can you upload the log here? So we can have a closer look at these errors. This bit of the code hasn't really been touched for a while so the patch should be ok on 2.3 as well.

Vrihub, 2024-12-20:
> Can you upload the log here? So we can have a closer look at these errors. This bit of the code hasn't really been touched for a while so the patch should be ok on 2.3 as well.

Here you are.

[blueman.log](https://github.com/user-attachments/files/18214780/blueman.log)

Vrihub, 2024-12-20:
Maybe this is useful too, for historical reference:
https://github.com/blueman-project/blueman/pull/263

infirit, 2024-12-20:
I suck. I'll update the PR in the next 10 minutes. I added an extra `shutil.move(src, dest)` by accident. Obvious the second time we run this the original file is gone.

Vrihub, 2024-12-20:
> edit: PRs have been updated

Thanks! Just tested the new version. Now everything works as expected, no side effects.

PR #2575 (2024-12-21): Revert "TransferService: small code optimization for on_transfer_comp…
This completely ignored the new filename we generated. fixes #2574
closed 2024-12-21 by infirit