blueman-project/blueman #2581: Pairing code disappears too quickly during device pairing

pback34, 2024-12-23:
blueman: 2.3.5
BlueZ: 5.64
Distribution: linux-mint Linux Mint 21.3
Desktop environment: X-Cinnamon

* [x ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

When attempting to pair a Bluetooth device using Blueman, the pairing code popup displayed on the screen disappears after only a few seconds. This behavior makes it challenging to complete the pairing process, especially when the user needs more time to enter the code on the device or confirm the pairing. This issue significantly impacts usability and may result in failed pairing attempts.

infirit, 2024-12-24:
Can you start blueman-applet described here https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman.

And then do the pair which has the issue. Then post it here.

cschramm, 2024-12-24:
blueman sends the notification with a timeout value of 0, meaning no timeout and removes it when bluetoothd sends a cancelation signal or another pairing. Iirc, cinnamon always shows notifications for a configured amount of time and then moves non-transient ones into its tray (unless configured to even discard them), no matter if a specific timeout, a default (-1), or no timeout (0) is requested. You should thus find the notification in your tray. That's the notification concept of your desktop.

infirit, 2024-12-24:
Ah, yeah it does indeed hide in the tray. An interesting design choice..

As this is intended behavior by Cinnamon I'll close this issue.

closed 2024-12-24 by infirit

pback34, 2024-12-24:
Ah, I see. Thank you!