blueman-project/blueman #2588: Blueman on Linux Mint does not request for a bluetooth code and connection is not made

Errie3, 2024-12-29:
blueman: 235
BlueZ:
Distribution:
Desktop environment: Clevo Linux Mint

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

--> Although Fairphone 5 G5 (on G2-4) is listed as Bluetooth apparatus and Fairphone gives a code to connect, Blueman does not give a request for a code and I can therefore not enter the code given by the android. Thus connection (linking) cannot be made and response from Fairphone after trying to link for a while is: code is incorrect.

Errie3, 2024-12-29:
Debug results Blueman:
blueman-applet --loglevel debug
blueman-applet 20.30.44 INFO     PluginManager:87 load_plugin: ['DhcpClient', 'StatusNotifierItem', 'KillSwitch', 'DiscvManager', 'StandardItems', 'NetUsage', 'ShowConnected', 'DisconnectItems', 'Networking', 'DBusService', 'Menu', 'NMDUNSupport', 'AutoConnect', 'GameControllerWakelock', 'NMPANSupport', 'RecentConns', 'StatusIcon', 'SerialManager', 'ConnectionNotifier', 'PowerManager', 'ExitItem', 'TransferService', 'AuthAgent', 'PPPSupport']
blueman-applet 20.30.44 WARNING  PluginManager:151 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 20.30.44 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 20.30.44 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 20.30.44 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '2C:6D:C1:28:55:B8', 'AddressType': 'public', 'Name': 'linuxlaptop', 'Alias': 'linuxlaptop', 'Class': 8126732, 'Powered': True, 'Discoverable': True, 'DiscoverableTimeout': 0, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-applet 20.30.44 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 20.30.44 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 20.30.44 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 20.30.44 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 20.30.44 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 20.30.44 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 20.30.44 WARNING  PluginManager:151 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 20.30.44 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 20.30.44 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 20.30.44 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 20.30.44 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 20.30.45 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 20.30.45 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 20.30.45 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 20.30.45 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 20.30.45 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 20.30.45 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 20.30.45 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 20.30.45 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 20.30.45 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 20.30.45 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 20.30.45 INFO     KillSwitch:106 io_event  : killswitch registered 2
blueman-applet 20.30.45 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 20.30.45 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 20.30.45 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 20.30.45 INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.80
blueman-applet 20.30.45 INFO     Applet:63 _on_dbus_name_appeared: org.bluez :1.3
blueman-applet 20.30.45 INFO     Functions:114 launch    : Gtk eventtime is 0, not using LaunchContext
[... cut ...]
blueman-applet 20.30.45 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem object at 0x7f87a36d4bb0>
blueman-applet 20.30.45 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <StatusIcon.StatusIcon object at 0x7f87a00ee280 (blueman+plugins+applet+StatusIcon+StatusIcon at 0x5652cc7878a0)>
blueman-applet 20.30.45 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.Menu.Menu object at 0x7f87a36d5060>

cschramm, 2024-12-30:
The log does not contain a pairing attempt and looks like the applet process immediately shut down after initialization. There probably was another blueman-applet process still running.

closed 2025-02-02 by github-actions[bot]