blueman-project/blueman #2621: Bug Report: GDBus.Error:org.freedesktop.DBus.Error.InvalidArgs: No such interface 'org.bluez.Battery1'

moltzaum, 2025-01-29:
Hello, I'm looking to file a bug report:

ubuntu version: 22.04.5 LTS
blueman version: 2.2.4-1
bluez version: 5.64-0ubuntu1.4 and 5.64-0ubuntu1 (both seem to be installed)

When connecting to JLAB Go keyboard via bluetooth, I get the following error:
```
blueman-manager 
blueman-manager 09.53.37 WARNING  ManagerDeviceList:341 _monitor_power_levels: Failed to get power levels, probably a LE device.
*** BUG ***
In pixman_region32_init_rect: Invalid rectangle passed
Set a breakpoint on '_pixman_log_error' to debug

*** BUG ***
In pixman_region32_init_rect: Invalid rectangle passed
Set a breakpoint on '_pixman_log_error' to debug

*** BUG ***
In pixman_region32_init_rect: Invalid rectangle passed
Set a breakpoint on '_pixman_log_error' to debug

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/bluez/Base.py", line 92, in get
    prop = self.call_sync(
gi.repository.GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.InvalidArgs: No such interface 'org.bluez.Battery1' (16)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 363, in _check_power_levels
    self._update_power_levels(tree_iter, device, cinfo)
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py", line 411, in _update_power_levels
    bars["battery"] = Battery(obj_path=device.get_object_path())["Percentage"]
  File "/usr/lib/python3/dist-packages/blueman/bluez/Base.py", line 135, in __getitem__
    return self.get(key)
  File "/usr/lib/python3/dist-packages/blueman/bluez/Base.py", line 106, in get
    raise parse_dbus_error(e)
blueman.bluez.errors.BluezDBusException: org.freedesktop.DBus.Error.InvalidArgs No such interface 'org.bluez.Battery1'
```

By editing the file `/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceList.py` to comment out line 441, I was able to connect the keyboard.

This was the change I made:
```
440      if ...
441          pass
442          # bars["battery"] = Battery(obj_path=device.get_object_path())["Percentage"]
```

I installed blueman because I could not connect using "Settings", which didn't give me any output or error message. That is to say: the inability for the JLab Go keyboard to connect to bluetooth initially doesn't specific to blueman. After commenting out the line and connecting successfully, the connection status is properly displayed in "Settings" and appears to work.

I'm not sure what action this requires, but I wanted to report it in case this finding could improve anything on this end.

infirit, 2025-01-30:
2.2.4 was released 3 years ago and much has changed and fixed since including this particular issue (iirc). You can try a git snapshot from https://blueman.cschramm.eu/debian/

moltzaum, 2025-01-31:
@infirit Pretty silly on my part then 😅. I upgraded Ubuntu to 24.04 and tried again, forgetting the keyboard and retrying. It worked with no issues. Thank you!

closed 2025-01-31 by moltzaum