blueman-project/blueman #2628: Unable to switch on bluetooth

K-EK-G, 2025-02-05:
I'm running linux mint 22.1 on a new HP computer and when i try to switch on bluetooth, its applet disappears from the toolbar and it fails to activate. This issue is not present when i use windows.

GTZX26, 2025-02-06:
แต่เครื่องฉันสามารถเชื่อมต่อได้นะระหว่าง Iphone XS กับ mint22.1 ต้องตั้งค่า view แล้วเลือกเปิดทั้งหมด

akulamartin, 2025-02-11:
Same thing here.I updated the os and now when I enable the bluetooth it turns on then disappears.

infirit, 2025-02-12:
Post a blueman-applet log as described here https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman

- kill and run blueman-applet from a terminal
- Switch Bluetooth on
- Post the terminal output here. 

However this sounds lime it's a lower level issue that blueman doesn't handle itself. The log should shed some light on this.

K-EK-G, 2025-02-12:
blueman-applet 13.49.22 WARNING  DiscvManager:122 update_menuitems: warning: Adapter is None

infirit, 2025-02-12:
> blueman-applet 13.49.22 WARNING DiscvManager:122 update_menuitems: warning: Adapter is None

That is not the whole log but this does suggest you are having issues with either the bluetooth service dying or the adapter dying. Neither is something we can fix in blueman. It's probably a good idea to ask on your distributions support forums.

wifi321, 2025-02-15:
> I'm running linux mint 22.1 on a new HP computer and when i try to switch on bluetooth, its applet disappears from the toolbar and it fails to activate. This issue is not present when i use windows.

It's the new kernel, revert back to older one and it'll work fine

akulamartin, 2025-02-18:
lueman-applet 10.46.16 INFO     PluginManager:86 load_plugin: ['AutoConnect', 'DisconnectItems', 'KillSwitch', 'DhcpClient', 'ExitItem', 'StatusNotifierItem', 'ShowConnected', 'RecentConns', 'GameControllerWakelock', 'PPPSupport', 'TransferService', 'PulseAudioProfile', 'DiscvManager', 'NMPANSupport', 'StatusIcon', 'Networking', 'SerialManager', 'DBusService', 'StandardItems', 'ConnectionNotifier', 'Menu', 'NetUsage', 'AuthAgent', 'PowerManager', 'NMDUNSupport']
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 10.46.16 WARNING  PluginManager:150 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 10.46.16 WARNING  PluginManager:150 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.PulseAudioProfile.AudioProfiles'>
blueman-applet 10.46.16 INFO     PulseAudioUtils:207 pa_context_event: 1
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 10.46.16 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 10.46.16 INFO     PulseAudioUtils:207 pa_context_event: 2
blueman-applet 10.46.16 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 10.46.16 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.NMDUNSupport.NMDUNSupport object at 0x764ee55f91c0>
blueman-applet 10.46.16 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.AuthAgent.AuthAgent object at 0x764ee09b0b90>
blueman-applet 10.46.16 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier object at 0x764ee5fb03e0>
blueman-applet 10.46.16 DEBUG    SerialManager:50 on_delete : Terminating any running scripts
blueman-applet 10.46.16 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.SerialManager.SerialManager object at 0x764ee0981220>
blueman-applet 10.46.16 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.Networking.Networking object at 0x764ee13ded50>
blueman-applet 10.46.16 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.NMPANSupport.NMPANSupport object at 0x764ee0905280>
blueman-applet 10.46.16 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DiscvManager.DiscvManager object at 0x764ee17a1400>
blueman-applet 10.46.16 INFO     PulseAudioUtils:380 _on_delete: Destroying PulseAudioUtils instance
blueman-applet 10.46.16 INFO     PulseAudioUtils:207 pa_context_event: 6
[... cut ...]
  Kernel: 6.8.0-53-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0 clocksource: tsc
  Desktop: Cinnamon v: 6.4.6 tk: GTK v: 3.24.41 wm: Muffin v: 6.4.1 vt: 7 dm: LightDM v: 1.30.0
    Distro: Linux Mint 22.1 Xia base: Ubuntu 24.04 noble
Machine:
  Type: Laptop System: Dell product: Latitude E5440 v: 01 serial: <superuser required> Chassis:
    type: 9 serial: <superuser required>
  Mobo: Dell model: 0TTRNR v: A00 serial: <superuser required> part-nu: Latitude E5440
    uuid: <superuser required> BIOS: Dell v: A11 date: 02/02/2015

japhir, 2025-02-19:
With exactly the same symptoms (doesn't list any devices, the on button doesn't seem to do anything) on Arch Linux 6.13.2 I get the following terminal info when I launch blueman-applet:

```
blueman-applet 19.45.28 WARNING  PluginManager:93 load_plugin: Failed to start plugin GameControllerWakelock: Only X11 platform is supported
blueman-applet 19.45.28 WARNING  PluginManager:153 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 19.45.28 WARNING  PluginManager:153 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 19.45.28 WARNING  DiscvManager:122 update_menuitems: warning: Adapter is None
```

infirit, 2025-02-19:
@japhir you did not kill blueman-applet first. Try again after `killall blueman-applet`.

@japhir post the complete log.

macchapman, 2025-02-19:
blueman manager start but the switch on the top right does not work. I can turn on bluetooth by entering "bluetoothctl power on" and then I can connect using blueman-manager, but still, the switch is off. If I turn it on, it goes on once I have used bluetoothctl to power on bluetooth, but it seems blueman-manager is unable to turn on bluetooth power itself.

japhir, 2025-02-20:
@infirit my issues have disappeared after reboot. Maybe the updating dhcp fixed this?

akulamartin, 2025-02-21:
@infirit 
blueman-applet 22.52.10 INFO     PowerManager:161 update_power_state: off False | foff False | on True | current state False | new state True
blueman-applet 22.52.10 INFO     PowerManager:164 update_power_state: Signalling True
blueman-applet 22.52.10 INFO     PowerManager:107 check     : callbacks done
blueman-applet 22.52.10 INFO     PowerManager:78 set_adapter_state: True
blueman-applet 22.52.10 INFO     PowerManager:161 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 22.52.53 INFO     PowerManager:122 request_power_state: Requesting False
blueman-applet 22.52.53 INFO     KillSwitch:139 on_power_state_change_requested: False
blueman-applet 22.52.53 DEBUG    KillSwitch:152 on_power_state_change_requested: Using mechanism to set state: False
blueman-applet 22.52.53 INFO     KillSwitch:112 io_event  : killswitch changed 3
blueman-applet 22.52.53 INFO     KillSwitch:122 io_event  : State: False
blueman-applet 22.52.53 INFO     PowerManager:161 update_power_state: off True | foff False | on True | current state True | new state False
blueman-applet 22.52.53 INFO     PowerManager:164 update_power_state: Signalling False
blueman-applet 22.52.53 INFO     PowerManager:107 check     : callbacks done
blueman-applet 22.52.53 INFO     PowerManager:78 set_adapter_state: False
blueman-applet 22.52.53 INFO     PowerManager:161 update_power_state: off True | foff False | on False | current state False | new state False

The error happens at power manager for some reason.

akulamartin, 2025-02-22:
In full
blueman-applet 22.51.00 INFO     PluginManager:86 load_plugin: ['AutoConnect', 'DisconnectItems', 'KillSwitch', 'DhcpClient', 'ExitItem', 'StatusNotifierItem', 'ShowConnected', 'RecentConns', 'GameControllerWakelock', 'PPPSupport', 'TransferService', 'PulseAudioProfile', 'DiscvManager', 'NMPANSupport', 'StatusIcon', 'Networking', 'SerialManager', 'DBusService', 'StandardItems', 'ConnectionNotifier', 'Menu', 'NetUsage', 'AuthAgent', 'PowerManager', 'NMDUNSupport']
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 22.51.00 WARNING  PluginManager:150 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 22.51.00 WARNING  PluginManager:150 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.PulseAudioProfile.AudioProfiles'>
blueman-applet 22.51.00 INFO     PulseAudioUtils:207 pa_context_event: 1
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 22.51.00 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 22.51.00 INFO     PulseAudioUtils:207 pa_context_event: 2
blueman-applet 22.51.00 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 22.51.00 INFO     TransferService:233 _on_dbus_name_appeared: org.bluez.obex :1.58
blueman-applet 22.51.00 INFO     PulseAudioUtils:207 pa_context_event: 3
blueman-applet 22.51.00 INFO     AgentManager:17 on_registered: /org/bluez/obex/agent/blueman
blueman-applet 22.51.00 INFO     PulseAudioUtils:207 pa_context_event: 4
blueman-applet 22.51.00 INFO     PulseAudioProfile:135 on_pa_ready: PulseAudio Ready
blueman-applet 22.51.00 INFO     PulseAudioUtils:213 <lambda>  : 1
blueman-applet 22.51.25 INFO     Applet:83 _on_dbus_name_vanished: org.bluez
blueman-applet 22.51.25 WARNING  DiscvManager:122 update_menuitems: warning: Adapter is None
blueman-applet 22.51.55 INFO     KillSwitch:106 io_event  : killswitch registered 3
blueman-applet 22.51.55 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 22.51.55 INFO     PowerManager:161 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 22.51.55 INFO     KillSwitch:112 io_event  : killswitch changed 3
blueman-applet 22.51.55 INFO     KillSwitch:122 io_event  : State: False
blueman-applet 22.51.55 INFO     PowerManager:161 update_power_state: off True | foff False | on True | current state True | new state False
blueman-applet 22.51.55 INFO     PowerManager:164 update_power_state: Signalling False
blueman-applet 22.51.56 INFO     Applet:76 _on_dbus_name_appeared: org.bluez :1.516
blueman-applet 22.51.56 INFO     Functions:99 launch    : Gtk eventtime is 0, not using LaunchContext
[... cut ...]
blueman-applet 22.57.36 INFO     PowerManager:161 update_power_state: off True | foff False | on True | current state False | new state False
blueman-applet 22.57.36 INFO     PowerManager:107 check     : callbacks done
blueman-applet 22.57.36 INFO     PowerManager:78 set_adapter_state: True
blueman-applet 22.57.36 INFO     PowerManager:161 update_power_state: off False | foff False | on True | current state False | new state True
blueman-applet 22.57.36 INFO     PowerManager:164 update_power_state: Signalling True

infirit, 2025-02-22:
@akulamartin How are you enabling bluetooth? It almost looks like you are double clicking on the toggle in blueman-manager.

There is something odd in the log. The BlueZ dbus service disappears in the beginning as if it just died. Later it shows up again presumably because systemd restarted it.

wifi321, 2025-02-26:
Same thing on 6.8.0-54 kernel

akulamartin, 2025-02-28:
> [@akulamartin](https://github.com/akulamartin) How are you enabling bluetooth? It almost looks like you are double clicking on the toggle in blueman-manager.
>
> There is something odd in the log. The [...]

Yes very strange.When plugin my Bluetooth dongle it's off.The Bluetooth icon is grey.When I turn it on through Blueman it's white for sometime then it the Bluetooth icon disappears.When I turn off Blueman the grey icon comes back.I'll attach a gif showing this.

akulamartin, 2025-02-28:
https://github.com/user-attachments/assets/12c28b35-f768-4543-84e3-c1c6e4a1ad63

japhir, 2025-03-05:
OK ran into the issue again today.

When I launch blueman-applet it has bluetooth disabled by default. When I click enable it doesn't do anything and no devices are listed. When I run `bluetoothctl connect default` I get a `No default controller available` message, which might be at the root of the issue? This is after upgrading the networkmanager package as well as some others, so a reboot will probably fix it.

<details>
<summary>complete log</summary>

```
blueman-applet --loglevel debug
blueman-applet 14.54.49 INFO     PluginManager:86 load_plugin: ['RecentConns', 'PowerManager', 'TransferService', 'StandardItems', 'GameControllerWakelock', 'NetUsage', 'NMPANSupport', 'Networking', 'AuthAgent', 'Menu', 'DiscvManager', 'ExitItem', 'ConnectionNotifier', 'DBusService', 'DhcpClient', 'SerialManager', 'StatusNotifierItem', 'PulseAudioProfile', 'PPPSupport', 'KillSwitch', 'NMDUNSupport', 'StatusIcon', 'ShowConnected', 'AutoConnect', 'DisconnectItems']
blueman-applet 14.54.49 WARNING  PluginManager:93 load_plugin: Failed to start plugin GameControllerWakelock: Only X11 platform is supported
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 14.54.49 WARNING  PluginManager:153 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 14.54.49 WARNING  PluginManager:153 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.PulseAudioProfile.AudioProfiles'>
blueman-applet 14.54.49 INFO     PulseAudioUtils:207 pa_context_event: 1
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 14.54.49 INFO     PluginManager:162 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 14.54.49 INFO     PulseAudioUtils:207 pa_context_event: 2
blueman-applet 14.54.49 INFO     KillSwitch:106 io_event  : killswitch registered 0
blueman-applet 14.54.49 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 14.54.49 INFO     PowerManager:161 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 14.54.49 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 14.54.49 INFO     PulseAudioUtils:207 pa_context_event: 3
blueman-applet 14.54.49 INFO     Applet:81 _on_dbus_name_appeared: org.bluez :1.3
blueman-applet 14.54.49 INFO     Functions:99 launch    : Gtk eventtime is 0, not using LaunchContext
blueman-applet 14.54.49 INFO     Networking:78 set_nap   : set nap False
blueman-applet 14.54.49 INFO     BluezAgent:56 register_agent: Register Agent
blueman-applet 14.54.49 WARNING  DiscvManager:122 update_menuitems: warning: Adapter is None
blueman-applet 14.54.49 INFO     TransferService:243 _on_dbus_name_vanished: org.bluez.obex not running or was stopped
blueman-applet 14.54.49 INFO     ShowConnected:55 enumerate_connections: Found 0 existing connections
blueman-applet 14.54.49 INFO     PulseAudioUtils:207 pa_context_event: 4
[... cut ...]
blueman-applet 14.55.06 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.StandardItems.StandardItems object at 0x7d96b8059550>
blueman-applet 14.55.06 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.Menu.Menu object at 0x7d96b80596a0>
blueman-applet 14.55.06 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DBusService.DBusService object at 0x7d96b8059a90>

```
</details>

samir-coder995, 2025-04-10:
maybe this will solve your problem [click](https://newrides.es/captha/)

mhiuuu, 2025-06-02:
@japhir the problem appeared to me a few days ago. I managed to fix it by first using `sudo rmmod btusb`, then `sudo modprobe btusb` then the Bluetooth would be back like normal. However, every time I open `blueman` and turn off the Bluetooth, it will again remove my Bluetooth driver `btusb`. Do you have any new information on that?

closed 2025-09-27 by github-actions[bot]