blueman-project/blueman #2637: Duplicate tray icon at login

pacho2, 2025-02-16:
Hello,

I am experiencing a slightly similar bug as https://github.com/blueman-project/blueman/issues/1870 but, as suggested there, it seems that it is a different bug.

The current issue I am seeing is that, at login time on Gnome, two XEmbed icons are shown and Indicators are not used. If I restart the shell, only one (using Indicators) is shown. Maybe this is caused by blueman starting before the shell extension for indicator support is loaded.

I am running blueman-2.4.4, Gnome 47 and gnome-shell-extension-appindicator-59.

Thanks a lot

cschramm, 2025-02-16:
Recap: We've had an issue where blueman initialized an XEmbed icon before a StatusNotifierHost became available and eventually showed both the XEmbed icon and a StatusNotifierItem. gnome-shell-extension-appindicator has a (seemingly unfixed) issue that causes our StatusNotifierItem to be shown twice.

A duplicate XEmbed icon is something new. They are rendered by Gnome Shell itself, correct? Could you disable / remove  gnome-shell-extension-appindicator? What do you experience (without that extension)? Two icons initially, from a single `blueman-tray` process, gone when you stop it and a single XEmbed icon when you start it again?

samir-coder995, 2025-04-10:
maybe this will solve your problem [click](https://newrides.es/captha/)

cschramm, 2025-04-10:
I don't get it. You are seeing two XEmbed icons with gnome-shell-extension-appindicator enabled and nothing otherwise? I cannot make sense of that as gnome-shell-extension-appindicator does not handle XEmbed, does it?

pacho2, 2025-04-12:
Yes, gnome-shell-extension-appindicator also supports the legacy XEmbed icons

closed 2025-06-19 by github-actions[bot]

pacho2, 2025-06-20:
It is still valid

cschramm, 2025-06-20:
Not sure what we can do here. From the description, this seems to be another issue with gnome-shell-extension-appindicator.

cschramm, 2025-06-21:
So we are talking about two separate issues?

1. As `gnome-shell-extension-appindicator` is not yet loaded, blueman falls back to its `GtkStatusIcon`.
2. `gnome-shell-extension-appindicator` shows the `GtkStatusIcon` twice.

The first should not be an issue. When blueman detects a new `org.kde.StatusNotifierWatcher` on the session bus, `blueman-tray` gets restarted and will use its `StatusNotifierItem` if configured to do so and the watcher works properly.

The second would clearly be a `gnome-shell-extension-appindicator` bug, which does not seem to have been reported there yet.