blueman-project/blueman #2644: I can't connect to my earbuds no matter what I do

cmhardy, 2025-02-23:
I'm running Linux Mint 22.1 and no matter what online fix I try, I just can't get it to connect.
My system info:

System:
  Kernel: 6.11.0-17-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0 clocksource: tsc
  Desktop: Cinnamon v: 6.4.7 tk: GTK v: 3.24.41 wm: Muffin v: 6.4.1 vt: 7 dm: LightDM v: 1.30.0
    Distro: Linux Mint 22.1 Xia base: Ubuntu 24.04 noble
Machine:
  Type: Desktop System: Dell product: OptiPlex 7050 v: N/A serial: <superuser required> Chassis:
    type: 3 serial: <superuser required>
  Mobo: Dell model: 0NW6H5 v: A00 serial: <superuser required> part-nu: 07A1
    uuid: <superuser required> UEFI: Dell v: 1.27.0 date: 09/18/2023
CPU:
  Info: quad core model: Intel Core i7-6700 bits: 64 type: MT MCP smt: enabled arch: Skylake-S
    rev: 3 cache: L1: 256 KiB L2: 1024 KiB L3: 8 MiB
  Speed (MHz): avg: 800 min/max: 800/3400 cores: 1: 800 2: 800 3: 800 4: 800 5: 800 6: 800 7: 800
    8: 800 bogomips: 54398
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel HD Graphics 530 vendor: Dell driver: i915 v: kernel arch: Gen-9 ports:
    active: DP-1,HDMI-A-1 empty: DP-2, DP-3, HDMI-A-2, HDMI-A-3 bus-ID: 00:02.0 chip-ID: 8086:1912
    class-ID: 0300
  Display: x11 server: X.Org v: 21.1.11 with: Xwayland v: 23.2.6 driver: X: loaded: modesetting
    unloaded: fbdev,vesa dri: iris gpu: i915 display-ID: :0 screens: 1
  Screen-1: 0 s-res: 2960x1050 s-dpi: 96 s-size: 783x278mm (30.83x10.94") s-diag: 831mm (32.71")
  Monitor-1: DP-1 pos: primary,left model: HP w2007 serial: <filter> res: 1680x1050 hz: 60
    dpi: 99 size: 433x271mm (17.05x10.67") diag: 511mm (20.1") modes: max: 1680x1050 min: 720x400
  Monitor-2: HDMI-A-1 mapped: HDMI-1 pos: right model-id: AGO 0x6201 serial: <filter>
    res: 1280x1024 dpi: 127 size: 256x192mm (10.08x7.56") diag: 378mm (14.9") modes: max: 1024x768
    min: 720x480
  API: EGL v: 1.5 hw: drv: intel iris platforms: device: 0 drv: iris device: 1 drv: swrast gbm:
    drv: iris surfaceless: drv: iris x11: drv: iris inactive: wayland
  API: OpenGL v: 4.6 compat-v: 4.5 vendor: intel mesa v: 24.2.8-1ubuntu1~24.04.1 glx-v: 1.4
    direct-render: yes renderer: Mesa Intel HD Graphics 530 (SKL GT2) device-ID: 8086:1912
Audio:
  Device-1: Intel 200 Series PCH HD Audio vendor: Dell driver: snd_hda_intel v: kernel
    bus-ID: 00:1f.3 chip-ID: 8086:a2f0 class-ID: 0403
  API: ALSA v: k6.11.0-17-generic status: kernel-api
  Server-1: PipeWire v: 1.0.5 status: active with: 1: pipewire-pulse status: active
    2: wireplumber status: active 3: pipewire-alsa type: plugin
Network:
  Device-1: Intel Ethernet I219-LM vendor: Dell driver: e1000e v: kernel port: N/A bus-ID: 00:1f.6
    chip-ID: 8086:15e3 class-ID: 0200
  IF: enp0s31f6 state: up speed: 100 Mbps duplex: full mac: <filter>
  Device-2: Realtek RTL8852BE PCIe 802.11ax Wireless Network driver: rtw89_8852be v: kernel pcie:
    speed: 2.5 GT/s lanes: 1 port: e000 bus-ID: 02:00.0 chip-ID: 10ec:b852 class-ID: 0280
  IF: wlp2s0 state: down mac: <filter>
  IF-ID-1: wlxe84e06adfd6b state: down mac: <filter>
Bluetooth:
  Device-1: Realtek 802.11ac NIC driver: btusb,rtw_8821cu type: USB rev: 2.0 speed: 480 Mb/s
    lanes: 1 bus-ID: 1-3:2 chip-ID: 0bda:c820 class-ID: e001 serial: <filter>
  Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter> bt-v: 4.2 lmp-v: 8 sub-v: f098
    hci-v: 8 rev: 75b8 class-ID: 7c0104
Drives:
  Local Storage: total: 1.82 TiB used: 292.5 GiB (15.7%)
  ID-1: /dev/nvme0n1 vendor: Western Digital model: WD BLACK SN770 1TB size: 931.51 GiB
    speed: 63.2 Gb/s lanes: 4 tech: SSD serial: <filter> fw-rev: 731030WD temp: 39.9 C scheme: GPT
  ID-2: /dev/sda vendor: Western Digital model: WD Blue SA510 2.5 1000GB size: 931.51 GiB
    speed: 6.0 Gb/s tech: SSD serial: <filter> fw-rev: 0100 scheme: GPT
Partition:
  ID-1: / size: 915.72 GiB used: 292.46 GiB (31.9%) fs: ext4 dev: /dev/sda2
  ID-2: /boot/efi size: 96 MiB used: 43.4 MiB (45.2%) fs: vfat dev: /dev/sda1
Swap:
  ID-1: swap-1 type: file size: 2 GiB used: 0 KiB (0.0%) priority: -2 file: /swapfile
USB:
  Hub-1: 1-0:1 info: hi-speed hub with single TT ports: 16 rev: 2.0 speed: 480 Mb/s lanes: 1
    chip-ID: 1d6b:0002 class-ID: 0900
  Device-1: 1-3:2 info: Realtek 802.11ac NIC type: bluetooth driver: btusb,rtw_8821cu
    interfaces: 3 rev: 2.0 speed: 480 Mb/s lanes: 1 power: 500mA chip-ID: 0bda:c820 class-ID: e001
    serial: <filter>
  Hub-2: 1-5:3 info: Terminus Hub ports: 4 rev: 2.0 speed: 480 Mb/s lanes: 1 power: 100mA
    chip-ID: 1a40:0101 class-ID: 0900
  Device-1: 1-6:4 info: SONiX AMAZON MD005 Wired Keyboard type: keyboard,HID
    driver: hid-generic,usbhid interfaces: 2 rev: 2.0 speed: 1.5 Mb/s lanes: 1 power: 100mA
    chip-ID: 413d:2114 class-ID: 0300
  Device-2: 1-8:5 info: Logitech Optical Wheel Mouse type: mouse driver: hid-generic,usbhid
    interfaces: 1 rev: 2.0 speed: 1.5 Mb/s lanes: 1 power: 100mA chip-ID: 046d:c016 class-ID: 0301
  Hub-3: 2-0:1 info: super-speed hub ports: 10 rev: 3.0 speed: 5 Gb/s lanes: 1 chip-ID: 1d6b:0003
    class-ID: 0900
Sensors:
  System Temperatures: cpu: 28.0 C mobo: N/A
  Fan Speeds (rpm): N/A
Repos:
  Packages: 2430 pm: dpkg pkgs: 2398 pm: flatpak pkgs: 32
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/google-chrome.list
    1: deb [arch=amd64] https: //dl.google.com/linux/chrome/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com xia main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu noble main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu noble-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu noble-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ noble-security main restricted universe multiverse
  Active apt repos in: /etc/apt/sources.list.d/opera-stable.list
    1: deb https: //deb.opera.com/opera-stable/ stable non-free
  No active apt repos in: /etc/apt/sources.list.d/steam-beta.list
  Active apt repos in: /etc/apt/sources.list.d/steam-stable.list
    1: deb [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
    2: deb-src [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
Info:
  Memory: total: 32 GiB available: 31.21 GiB used: 3.76 GiB (12.1%)
  Processes: 275 Power: uptime: 8m states: freeze,mem,disk suspend: deep wakeups: 0
    hibernate: platform Init: systemd v: 255 target: graphical (5) default: graphical
  Compilers: gcc: 13.3.0 Client: Cinnamon v: 6.4.7 inxi: 3.3.34

sujaldey004, 2025-02-23:
same goes for me

anuptiwari17, 2025-02-24:
exactly same problem, do anyone knows how to resolve this?