blueman-project/blueman #2649: Linux Mint Cinnamon Edition - Bluetooth Issue

jud15411, 2025-02-27:
I am trying to pair my Bluetooth headphones to my Linux laptop so I can get some independent work done. The headphones will show up on my phone but when i go over to the Bluetooth manager on Linux, the headphones will not show up, no matter how many times I search for them. I already tried removing the bluez Bluetooth manager and reinstalling that along side with the blue-man. Neither of these worked for my case. I still can not find the headphones. Any suggestions would be greatly appreciated!

Samuelfcampbell2003, 2025-02-27:
Man I feel your pain!  Been through the same thing.
Come to find out for me It dosent matter Samsung Windows MackBook Pro, 808 Speaker, Pure Boom, Altech Speaker, or just about any device, Interfaces are easy to pick up on just try walking between a connected device you will be able to hear it sa the transmission is cut by your body. Devices too close to previous paired to a device etc... or cross transmitting waves. Some ti.es I have to shut bown all other possible devices. Then your over half the way their. Then check to see if the devise your search is actually ready to be paired or connected. Just because the device is on is sometimes not enough. It needs to be flash omitting an appropriate color usually flashing green light some where. After it is through pairing and becomes connected it will usually turn to a solid Blue light. Try one devise at a time. Hope this brings you as much joy and satisfaction as it has me. Good Luck!
~ Samuel F Campbell 🙂‍↕️

TGinAZ, 2025-03-01:
I just installed 22.1 on an old Lenovo YOGA 900 and even after trying removing and installing bluez, the Bluetooth Manager keeps the search greyed out.... I can't even do a search, let alone connect my earbuds.

(Before removing Windows 10, I could connect the earbuds to this system, so I know the equipment is functional.)

Anyone have an other suggestions?