blueman-project/blueman #2669: Keyboard insanity

EdmundDuckDuck, 2025-03-12:
Connecting works
Pairing fails every time
Sometimes I get a PIN number with no place to type it in
Other times I get a windows for a PIN code and get no pin code.

dbass81, 2025-03-15:
Typically when pairing a Bluetooth Keyboard, you type the PIN and hit enter, which should pair the two.

EdmundDuckDuck, 2025-03-15:
Thanks, eventually it paired.
I did not read anywhere how it was supposed to work an don't understand why I sometimes had a window, presumable to put in the pin.
Anyway it works now, thanks

dbass81, 2025-03-15:
> Thanks, eventually it paired. I did not read anywhere how it was supposed to work an don't understand why I sometimes had a window, presumable to put in the pin. Anyway it works now, thanks

Bluetooth on Linux seems broken, or rather it doesn't work right 100% of the time.  I removed Blueman, and I am solely using bluez-firmware on Debian Bookworm, with bluetoothctl on the CLI.