blueman-project/blueman #2676: Bluetooth does not work after turning it off and on again with the button

ernoone, 2025-03-19:
Bluetooth does not work after turning it off and on again with the button 
The window with settings all settings buttons is disabled

[image]

tomcodes, 2025-03-20:
Same here since yesterday after an update of my Arch system.

Competitive-Dev, 2025-03-24:
same issue with me in linux mint 22, { I am new to linux btw }

cschramm, 2025-03-24:
Please run `bluetoothctl` in one terminal window and `rfkill event` in another and see what the button does.

If the `PowerManager` plugin is enabled (default), disabling it unpowers the adapter:

```
[CHG] Controller ... Powered: no
[CHG] Controller ... PowerState: off
```

If the `KillSwitch` plugin is enabled (default), it also soft-blocks the Bluetooth switch:

```
... type 2 op 2 soft 1 hard 0
```

```
[CHG] Controller ... PowerState: off-blocked
```

Enabling it does the opposite:

```
... type 2 op 2 soft 0 hard 0
```

```
[CHG] Controller ... PowerState: on
[CHG] Controller ... Powered: yes
```

(@infirit: By the way, if the plugins are disabled, the button does nothing 🙈. It should probably not even be available then.)

infirit, 2025-03-24:
Oops 😅, I'll fix that soon(ish).

samir-coder995, 2025-04-10:
maybe this will solve your problem [click](https://newrides.es/captha/)