blueman-project/blueman #2677: Add python3-gobject to Fedora Dependency

Tahinli, 2025-03-19:
Hi, I was having errors like:

```python
AttributeError: 'NoneType' object has no attribute 'surface'
Traceback (most recent call last):
  File "/usr/lib/python3.13/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 666, in _set_cell_data
    cell.set_property("surface", row["device_surface"].surface)
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'surface'
TypeError: Couldn't find foreign struct converter for 'cairo.Context'
```

I wasnt able to see my bluetooth device in blueman but it was a visual problem. I was able to click and do something but couldn't see it. After some research I understand that I needed to install "python3-gobject". Please define this package as a dependency for blueman in Fedora otherwise blueman is not working properly.

After installing python3-gobject, now it works.

Tahinli, 2025-03-21:
> We do have it listed in [DEPENDENCIES](https://github.com/blueman-project/blueman/blob/main/Dependencies.md), its PyGobject. However the error was more likely due to not having PyCairo as that is what provides the surface and Context objects
> 
> How did you install blueman on fedora?

```
sudo dnf install blueman
```

I did it several times but until I install pygobject via "sudo dnf install python3-gobject" it wasn't working properly as I said.

cschramm, 2025-03-21:
Hi, you need to file a bug at https://bugzilla.redhat.com/ for issues with the Fedora package.

In its history I see [this suspicious change](https://src.fedoraproject.org/rpms/blueman/c/8d776efb0b3a3997d48b6c8a5e2cc7dcaed05df4) from `pygobject` to its subpackage `python3-gobject-base`. It is described to provide "the non-cairo specific bits", so it probably should have been `python3-gobject` instead. This was eventually changed [here](https://src.fedoraproject.org/rpms/blueman/c/3db80f1e80388869e8bd1665e57a3f9fb6de0dbd) to `python3dist(pygobject)`. I have no idea what that resolves to, but I assume it's still the base package for some reason.

Tahinli, 2025-03-21:
I reported, I hope in a correct way: https://bugzilla.redhat.com/show_bug.cgi?id=2354051

suve, 2025-03-21:
```
$ rpm -q --provides python3-gobject
python-gobject = 3.50.0-3.fc42
python3-gobject = 3.50.0-3.fc42
python3-gobject(x86-64) = 3.50.0-3.fc42
python3.13-gobject = 3.50.0-3.fc42
```
Hmmm, so `python3-gobject` does not provide `python3dist(pygobject)`.

```
$ rpm -q --provides python3-gobject-base
python-gobject-base = 3.50.0-3.fc42
python3-gobject-base = 3.50.0-3.fc42
python3-gobject-base(x86-64) = 3.50.0-3.fc42
python3-gobject-base-noarch = 3.50.0-3.fc42
python3.13-gobject-base = 3.50.0-3.fc42
python3.13dist(pygobject) = 3.50
python3dist(pygobject) = 3.50
```
It is provided by `python3-gobject-base`, and that's what `blueman` pulls in.

```
$ rpm -qi python3-gobject-base
Name        : python3-gobject-base
Version     : 3.50.0
Release     : 3.fc42
Architecture: x86_64
Install Date: Tue 18 Mar 2025 16:56:33 CET
Group       : Unspecified
Size        : 1579394
License     : LGPL-2.1-or-later
Signature   : RSA/SHA256, Sat 18 Jan 2025 13:19:14 CET, Key ID c8ac4916105ef944
Source RPM  : pygobject3-3.50.0-3.fc42.src.rpm
Build Date  : Sat 18 Jan 2025 12:18:27 CET
Build Host  : buildvm-x86-09.iad2.fedoraproject.org
Packager    : Fedora Project
Vendor      : Fedora Project
URL         : https://wiki.gnome.org/Projects/PyGObject
Bug URL     : https://bugz.fedoraproject.org/pygobject3
Summary     : Python 3 bindings for GObject Introspection base package
Description : This package provides the non-cairo specific bits of the GObject
              Introspection library.
```
Welp, "non-cairo specific bits". I'd very much prefer to keep using `python3dist()` for Python dependencies, but looks like in this case, it must be `python3-gobject`.

infirit, 2025-03-21:
Thanks @suve for the quick action.

@Tahinli not sure how long it takes for the update propagate to the users but it should be fixed now.

closed 2025-03-21 by infirit

Tahinli, 2025-03-30:
Happy news!

[image]