blueman-project/blueman #2678: Stopped seeing the Bluetooth devices

ulalum, 2025-03-21:
I did not do any changes, but it can't find and connect the Bluetooth JBL speaker I have used with it already. What can be the issue?

`sudo systemctl status Bluetooth

sudo systemctl status bluetooth
[sudo] пароль до user:          
● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor pre>
     Active: active (running) since Fri 2025-03-21 13:10:24 EET; 10min ago
       Docs: man:bluetoothd(8)
   Main PID: 710 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 9211)
     Memory: 2.2M
        CPU: 60ms
     CGroup: /system.slice/bluetooth.service
             └─710 /usr/lib/bluetooth/bluetoothd
`

sirenwerks, 2025-03-24:
Same here.

System:
  Kernel: 6.8.0-55-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0
  Desktop: Cinnamon v: 6.4.8 tk: GTK v: 3.24.41 wm: Muffin dm: LightDM
    Distro: Linux Mint 22.1 Xia base: Ubuntu 24.04 noble
Machine:
  Type: Mini-pc Mobo: AZW model: SER v: V2.0 serial: <superuser required> UEFI: American
    Megatrends LLC. v: SER5H512 date: 09/06/2024
CPU:
  Info: 8-core model: AMD Ryzen 7 PRO 5850U with Radeon Graphics bits: 64 type: MT MCP arch: Zen 3
    rev: 0 cache: L1: 512 KiB L2: 4 MiB L3: 16 MiB
  Speed (MHz): avg: 1160 high: 2423 min/max: 400/4507 cores: 1: 1994 2: 1914 3: 1928 4: 400
    5: 400 6: 400 7: 400 8: 400 9: 400 10: 2394 11: 2423 12: 2395 13: 400 14: 400 15: 400 16: 1916
    bogomips: 60687
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Graphics:
  Device-1: AMD Cezanne [Radeon Vega Series / Radeon Mobile Series] driver: amdgpu v: kernel
    arch: GCN-5 pcie: speed: 8 GT/s lanes: 16 ports: active: HDMI-A-1 empty: DP-1,DP-2,DP-3
    bus-ID: 04:00.0 chip-ID: 1002:1638 temp: 46.0 C
  Display: x11 server: X.Org v: 21.1.11 with: Xwayland v: 23.2.6 driver: X: loaded: amdgpu
    unloaded: fbdev,modesetting,vesa dri: radeonsi gpu: amdgpu display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1920x1080 s-dpi: 96
  Monitor-1: HDMI-A-1 mapped: HDMI-A-0 model: Dell P2417H res: 1920x1080 dpi: 93
    diag: 604mm (23.8")
  API: EGL v: 1.5 platforms: device: 0 drv: radeonsi device: 1 drv: swrast gbm: drv: kms_swrast
    surfaceless: drv: radeonsi x11: drv: radeonsi inactive: wayland
  API: OpenGL v: 4.6 compat-v: 4.5 vendor: amd mesa v: 24.2.8-1ubuntu1~24.04.1 glx-v: 1.4
    direct-render: yes renderer: AMD Radeon Graphics (radeonsi renoir LLVM 19.1.1 DRM 3.57
    6.8.0-55-generic) device-ID: 1002:1638
Audio:
  Device-1: AMD Renoir Radeon High Definition Audio driver: snd_hda_intel v: kernel pcie:
    speed: 8 GT/s lanes: 16 bus-ID: 04:00.1 chip-ID: 1002:1637
  Device-2: AMD ACP/ACP3X/ACP6x Audio Coprocessor driver: snd_rn_pci_acp3x v: kernel pcie:
    speed: 8 GT/s lanes: 16 bus-ID: 04:00.5 chip-ID: 1022:15e2
  Device-3: AMD Family 17h/19h HD Audio driver: snd_hda_intel v: kernel pcie: speed: 8 GT/s
    lanes: 16 bus-ID: 04:00.6 chip-ID: 1022:15e3
  API: ALSA v: k6.8.0-55-generic status: kernel-api
  Server-1: PipeWire v: 1.0.5 status: active with: 1: pipewire-pulse status: active
    2: wireplumber status: active 3: pipewire-alsa type: plugin
Network:
  Device-1: Realtek RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet driver: r8169 v: kernel
    pcie: speed: 2.5 GT/s lanes: 1 port: f000 bus-ID: 01:00.0 chip-ID: 10ec:8168
  IF: enp1s0 state: up speed: 1000 Mbps duplex: full mac: <filter>
  Device-2: Intel Wi-Fi 6 AX200 driver: iwlwifi v: kernel pcie: speed: 5 GT/s lanes: 1
    bus-ID: 02:00.0 chip-ID: 8086:2723
  IF: wlo1 state: down mac: <filter>
  IF-ID-1: nordlynx state: unknown speed: N/A duplex: N/A mac: N/A
Bluetooth:
  Device-1: Intel AX200 Bluetooth driver: btusb v: 0.8 type: USB rev: 2.0 speed: 12 Mb/s lanes: 1
    bus-ID: 3-3:3 chip-ID: 8087:0029
  Report: hciconfig ID: hci0 rfk-id: 0 state: down bt-service: enabled,running rfk-block:
    hardware: no software: no address: <filter>
Drives:
  Local Storage: total: 10.01 TiB used: 1.32 TiB (13.2%)
  ID-1: /dev/nvme0n1 vendor: Crucial model: CT1000P3PSSD8 size: 931.51 GiB speed: 63.2 Gb/s
    lanes: 4 serial: <filter> temp: 43.9 C
  ID-2: /dev/sda vendor: Seagate model: Ultra Slim MT size: 1.82 TiB type: USB rev: 3.0
    spd: 5 Gb/s lanes: 1 serial: <filter>
  ID-3: /dev/sdb vendor: Seagate model: Backup+ Hub BK size: 7.28 TiB type: USB rev: 2.1
    spd: 480 Mb/s lanes: 1 serial: <filter>
Partition:
  ID-1: / size: 485.74 GiB used: 460.98 GiB (94.9%) fs: ext4 dev: /dev/nvme0n1p5
  ID-2: /boot/efi size: 96 MiB used: 33.5 MiB (34.9%) fs: vfat dev: /dev/nvme0n1p1
Swap:
  ID-1: swap-1 type: file size: 2 GiB used: 2 GiB (100.0%) priority: -2 file: /swapfile
Sensors:
  System Temperatures: cpu: 55.2 C mobo: N/A gpu: amdgpu temp: 46.0 C
  Fan Speeds (rpm): N/A
Repos:
  Packages: 2985 pm: dpkg pkgs: 2949 pm: flatpak pkgs: 32 pm: snap pkgs: 4
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/flacon-ppa-noble.list
    1: deb [signed-by=/etc/apt/keyrings/flacon-ppa-noble.gpg] https: //ppa.launchpadcontent.net/flacon/ppa/ubuntu noble main
  Active apt repos in: /etc/apt/sources.list.d/google-chrome.list
    1: deb [arch=amd64] https: //dl.google.com/linux/chrome/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/google-earth-pro.list
    1: deb [arch=amd64] http: //dl.google.com/linux/earth/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/insync.list
    1: deb [signed-by=/etc/apt/trusted.gpg.d/insynchq.gpg] http: //apt.insync.io/mint wilma non-free contrib
  Active apt repos in: /etc/apt/sources.list.d/nordvpn.list
    1: deb https: //repo.nordvpn.com/deb/nordvpn/debian stable main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com xia main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu noble main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu noble-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu noble-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ noble-security main restricted universe multiverse
  Active apt repos in: /etc/apt/sources.list.d/signal-xenial.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/signal-desktop-keyring.gpg] https: //updates.signal.org/desktop/apt xenial main
Info:
  Memory: total: 28 GiB available: 27.33 GiB used: 2.91 GiB (10.7%)
  Processes: 419 Power: uptime: 2d 15h 56m wakeups: 0 Init: systemd v: 255 target: graphical (5)
    default: graphical
  Compilers: gcc: 13.3.0 Client: Unknown python3.12 client inxi: 3.3.34

MarcosMoraes23, 2025-03-24:
Same issue found here. 

 bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset>
     Active: active (running) since Mon 2025-03-24 12:25:26 -03; 20min ago
       Docs: man:bluetoothd(8)
   Main PID: 4614 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 37468)
     Memory: 984.0K (peak: 1.6M)
        CPU: 40ms
     CGroup: /system.slice/bluetooth.service
             └─4614 /usr/libexec/bluetooth/bluetoothd

humuraelvin, 2025-03-31:
Try to run systemctl restart dbus system

sirenwerks, 2025-04-01:
> Try to run systemctl restart dbus system

Tried this and ended up with a blueman-applet error message window on start-up that stated:

> You might not be able to connect to the Bluetooth network via this machine
> 
> Exception
> g-io-error-quark: GDBus.Error:org.blueman.Error: Not authorized (36)

I've tried re-installing and scrubbing numerous times. This Beelink mini worked flawlessly until the last Mint update, even after it worked briefly and then failed.  The Device tab in the Bluetooth Devices window is always grayed out.

Bryan

altr-r, 2025-04-04:
same here tried everything on the internet now clicked the report a problem section and saw your post

samir-coder995, 2025-04-10:
maybe this will solve your problem [click](https://newrides.es/captha/)