blueman-project/blueman #2692: All buttons appear disabled

XenamAnara, 2025-04-06:
Despite being able to turn it on, I can't press any button as they are all grey and disabled

naveenkumar-42, 2025-04-08:
yeah me too have same issue

Codechilla, 2025-04-09:
Same here. I'm on Linux Mint. Seems to be intermittent and rare, but a real pain when it happens. Each time it happens, I have to tinker for a while to get BT working again. Been sporadically happening for a couple months.

What seems to work for me sometimes is to kill all power (i.e. unplug), and remove the BT device. Replace the device, turn on power, and boot.

I have an Asus TUF x570 Pro WiFi MoBo. Not sure what is causing it, but never had any problems with BT on Windows on the same machine. Not that I like Windows or anything...just using as reference.

samir-coder995, 2025-04-10:
maybe this will solve your problem [click](https://newrides.es/captha/)

Competitive-Dev, 2025-04-13:
Result For :- **rfkill**
```
ID TYPE      DEVICE              SOFT      HARD
 0 wlan      asus-wlan      unblocked unblocked
 1 bluetooth asus-bluetooth unblocked unblocked
 2 bluetooth hci0           unblocked unblocked
 4 wlan      phy1           unblocked unblocked
```

Result for :- **bluetoothctl power on**
`No default controller available
`

SHirakoTTeku, 2025-05-01:
> If no controller is available (and not killswitch is blocked), there is nothing that blueman can do for you.

don't reply to people if you don't know what the solution is, This issue has been persistent for weeks for alot of users, This is because of outdated updates, I already checked youtube for solutions and there is 1 guy who managed to solve issue by updating the ubunto's  kernal to latest specific version. this is simply because Linux mint hasn't reached that update yet. and neither did blueman's plugins for Linux mint.

cschramm, 2025-05-02:
This is the blueman project. This is neither a platform for Linux Mint nor Bluetooth drivers support.