blueman-project/blueman #2703: Adapters not shown

fil3000, 2025-04-18:
The menu point Adapter drops down with only 3 horizontal lines and the exit sign afterwards. I have checked the status & rights and reinstalled blueman. Was working fine with DS4 before.

SNOWCHILDwolf256, 2025-04-19:
I have the same exact problem and did exactly that. However, upon restarting the device the adapters appear and the search button works but still no other nearby bluetooth devices were scanned. Also, the adapter later disappeared again and the three  lines were back.

PeterHaun, 2025-04-19:
Same here. Bluetooth Icon disappears after 2-3 seconds from the taskbar.

cschramm, 2025-04-21:
Sounds like your system does not provide an adapter (check `bluetoothctl list`). If there is no adapter and no soft-blocked killswitch (check `rfkill`) that could hide one, there is nothing that blueman can do for you.