blueman-project/blueman #2712: issue

Royaluser-cloud, 2025-04-27:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/main/PluginManager.py", line 62, in load_plugin
    self.__load_plugin(self.__classes[name])
  File "/usr/lib/python3/dist-packages/blueman/main/PluginManager.py", line 162, in __load_plugin
    inst._load()
  File "/usr/lib/python3/dist-packages/blueman/plugins/AppletPlugin.py", line 55, in _load
    self.on_manager_state_changed(self.parent.manager_state)
  File "/usr/lib/python3/dist-packages/blueman/plugins/applet/AuthAgent.py", line 22, in on_manager_state_changed
    self._agent.register_agent()
  File "/usr/lib/python3/dist-packages/blueman/main/applet/BluezAgent.py", line 57, in register_agent
    self.register()
  File "/usr/lib/python3/dist-packages/blueman/main/DbusService.py", line 114, in register
    regid = self._bus.register_object(
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
gi.repository.GLib.GError: g-io-error-quark: An object is already exported for the interface org.bluez.Agent1 at /org/bluez/agent/blueman (2)

Amiurion, 2025-04-28:
Hi ,
I'm running newly installed Linux Mint 22.1 
I'm also having problems with connecting / pairing to Bluetooth devices. I am getting similar message/s. I hope this might help to solve the issue: 

Failed to connect: org.bluez.Error.NotAvailable br-connection-profile-unavailable

Connection Failed: File "/usr/lib/python3/dist-packages/blueman/bluez/Base.py", line 100, in get
    prop = self.__proxy.call_sync(
           ^^^^^^^^^^^^^^^^^^^^^^^
gi.repository.GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.UnknownObject: Method "Get" with signature "ss" on interface "org.freedesktop.DBus.Properties" doesn't exist
 (41)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/main/DbusService.py", line 154, in _handle_method_call
    method(*(args + (ok, lambda exception: self._return_dbus_error(invocation, exception))))
  File "/usr/lib/python3/dist-packages/blueman/plugins/applet/DBusService.py", line 72, in connect_service
    self.parent.Plugins.RecentConns.notify(object_path, uuid)
  File "/usr/lib/python3/dist-packages/blueman/plugins/applet/RecentConns.py", line 144, in notify
    self._rebuild()
  File "/usr/lib/python3/dist-packages/blueman/plugins/applet/RecentConns.py", line 78, in _rebuild
    items = self._get_items()
            ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/blueman/plugins/applet/RecentConns.py", line 206, in _get_items
    return sorted(
           ^^^^^^^
  File "/usr/lib/python3/dist-packages/blueman/plugins/applet/RecentConns.py", line 219, in <genexpr>
    if (device := self._get_device_path(i["adapter"], i["address"]))),
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/blueman/plugins/applet/RecentConns.py", line 202, in _get_device_path
    device = self.parent.Manager.find_device(address, adapter.get_object_path())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/blueman/bluez/Manager.py", line 146, in find_device
    if device['Address'] == address:
       ~~~~~~^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/blueman/bluez/Base.py", line 150, in __getitem__
    return self.get(key)
           ^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/blueman/bluez/Base.py", line 114, in get
    raise parse_dbus_error(e)
blueman.bluez.errors.BluezDBusException: org.freedesktop.DBus.Error.UnknownObject Method "Get" with signature "ss" on interface "org.freedesktop.DBus.Properties" doesn't exist

crazybee83, 2025-09-18:
I am having a problem getting my Android Moto g5 to connect to my pc. Im trying to get kde connect to work. Any help would be appreciated. I am also new to linux in general so I dont know how to find the problem or anything related to finding the problem