blueman-project/blueman #2713: Device tab grayed out and no devices show

sirenwerks, 2025-04-28:
I;m running Linux Mint and about three weeks ago Bluetooth went belly up. The Device tab is grayed out and not amount of turning Bluetooth on and off or un/re-installing fixes the issue. Help. I'm tired of not being able to use this machine without audio.

Bryan

System:
  Kernel: 6.8.0-58-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0 clocksource: tsc
  Desktop: Cinnamon v: 6.4.8 tk: GTK v: 3.24.41 wm: Muffin v: 6.4.1 vt: 7 dm: LightDM v: 1.30.0
    Distro: Linux Mint 22.1 Xia base: Ubuntu 24.04 noble
Machine:
  Type: Mini-pc Mobo: AZW model: SER v: V2.0 serial: <superuser required>
    uuid: <superuser required> UEFI: American Megatrends LLC. v: SER5H512 date: 09/06/2024
CPU:
  Info: 8-core model: AMD Ryzen 7 PRO 5850U with Radeon Graphics bits: 64 type: MT MCP smt: enabled
    arch: Zen 3 rev: 0 cache: L1: 512 KiB L2: 4 MiB L3: 16 MiB
  Speed (MHz): avg: 2074 high: 4393 min/max: 400/4507 cores: 1: 3652 2: 400 3: 400 4: 400 5: 400
    6: 400 7: 400 8: 400 9: 3515 10: 3500 11: 3514 12: 3513 13: 3514 14: 400 15: 4392 16: 4393
    bogomips: 60690
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Graphics:
  Device-1: AMD Cezanne [Radeon Vega Series / Radeon Mobile Series] driver: amdgpu v: kernel
    arch: GCN-5 pcie: speed: 8 GT/s lanes: 16 ports: active: DP-1,HDMI-A-1 empty: DP-2,DP-3
    bus-ID: 04:00.0 chip-ID: 1002:1638 class-ID: 0300 temp: 46.0 C
  Device-2: Logitech C920 PRO HD Webcam driver: snd-usb-audio,uvcvideo type: USB rev: 2.0
    speed: 480 Mb/s lanes: 1 bus-ID: 3-2.3.1.2:107 chip-ID: 046d:08e5 class-ID: 0102 serial: <filter>
  Display: x11 server: X.Org v: 21.1.11 with: Xwayland v: 23.2.6 driver: X: loaded: amdgpu
    unloaded: fbdev,modesetting,vesa dri: radeonsi gpu: amdgpu display-ID: :0 screens: 1
  Screen-1: 0 s-res: 3840x1080 s-dpi: 96 s-size: 1016x286mm (40.00x11.26")
    s-diag: 1055mm (41.55")
  Monitor-1: DP-1 mapped: DisplayPort-0 pos: primary,right model: Dell P2417H serial: <filter>
    res: 1920x1080 hz: 60 dpi: 93 size: 527x296mm (20.75x11.65") diag: 604mm (23.8") modes:
    max: 1920x1080 min: 720x400
  Monitor-2: HDMI-A-1 mapped: HDMI-A-0 pos: left model: Dell P2417H serial: <filter>
    res: 1920x1080 hz: 60 dpi: 93 size: 527x296mm (20.75x11.65") diag: 604mm (23.8") modes:
    max: 1920x1080 min: 720x400
  API: EGL v: 1.5 hw: drv: amd radeonsi platforms: device: 0 drv: radeonsi device: 1 drv: swrast
    gbm: drv: kms_swrast surfaceless: drv: radeonsi x11: drv: radeonsi inactive: wayland
  API: OpenGL v: 4.6 compat-v: 4.5 vendor: amd mesa v: 24.2.8-1ubuntu1~24.04.1 glx-v: 1.4
    direct-render: yes renderer: AMD Radeon Graphics (radeonsi renoir LLVM 19.1.1 DRM 3.57
    6.8.0-58-generic) device-ID: 1002:1638
Audio:
  Device-1: AMD Renoir Radeon High Definition Audio driver: snd_hda_intel v: kernel pcie:
    speed: 8 GT/s lanes: 16 bus-ID: 04:00.1 chip-ID: 1002:1637 class-ID: 0403
  Device-2: AMD ACP/ACP3X/ACP6x Audio Coprocessor driver: snd_rn_pci_acp3x v: kernel pcie:
    speed: 8 GT/s lanes: 16 bus-ID: 04:00.5 chip-ID: 1022:15e2 class-ID: 0480
  Device-3: AMD Family 17h/19h HD Audio driver: snd_hda_intel v: kernel pcie: speed: 8 GT/s
    lanes: 16 bus-ID: 04:00.6 chip-ID: 1022:15e3 class-ID: 0403
  Device-4: Logitech C920 PRO HD Webcam driver: snd-usb-audio,uvcvideo type: USB rev: 2.0
    speed: 480 Mb/s lanes: 1 bus-ID: 3-2.3.1.2:107 chip-ID: 046d:08e5 class-ID: 0102 serial: <filter>
  API: ALSA v: k6.8.0-58-generic status: kernel-api
  Server-1: PipeWire v: 1.0.5 status: active with: 1: pipewire-pulse status: active
    2: wireplumber status: active 3: pipewire-alsa type: plugin
Network:
  Device-1: Realtek RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet driver: r8169 v: kernel
    pcie: speed: 2.5 GT/s lanes: 1 port: f000 bus-ID: 01:00.0 chip-ID: 10ec:8168 class-ID: 0200
  IF: enp1s0 state: up speed: 1000 Mbps duplex: full mac: <filter>
  Device-2: Intel Wi-Fi 6 AX200 driver: iwlwifi v: kernel pcie: speed: 5 GT/s lanes: 1
    bus-ID: 02:00.0 chip-ID: 8086:2723 class-ID: 0280
  IF: wlo1 state: down mac: <filter>
  IF-ID-1: nordlynx state: unknown speed: N/A duplex: N/A mac: N/A
Bluetooth:
  Device-1: Intel AX200 Bluetooth driver: btusb v: 0.8 type: USB rev: 2.0 speed: 12 Mb/s lanes: 1
    bus-ID: 3-3:3 chip-ID: 8087:0029 class-ID: e001
  Report: hciconfig ID: hci0 rfk-id: 1 state: down bt-service: enabled,running rfk-block:
    hardware: no software: no address: <filter>
Drives:
  Local Storage: total: 10.01 TiB used: 626.65 GiB (6.1%)
  ID-1: /dev/nvme0n1 vendor: Crucial model: CT1000P3PSSD8 size: 931.51 GiB speed: 63.2 Gb/s
    lanes: 4 tech: SSD serial: <filter> fw-rev: P9CR413 temp: 42.9 C scheme: GPT
  ID-2: /dev/sdb vendor: Seagate model: Backup+ Hub BK size: 7.28 TiB type: USB rev: 2.1
    spd: 480 Mb/s lanes: 1 tech: N/A serial: <filter> fw-rev: D785 scheme: GPT
[... 68 lines cut ...]
  Packages: 3002 pm: dpkg pkgs: 2968 pm: flatpak pkgs: 30 pm: snap pkgs: 4
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/flacon-ppa-noble.list
    1: deb [signed-by=/etc/apt/keyrings/flacon-ppa-noble.gpg] https: //ppa.launchpadcontent.net/flacon/ppa/ubuntu noble main
  Active apt repos in: /etc/apt/sources.list.d/google-chrome.list
    1: deb [arch=amd64] https: //dl.google.com/linux/chrome/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/google-earth-pro.list
    1: deb [arch=amd64] http: //dl.google.com/linux/earth/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/insync.list
    1: deb [signed-by=/etc/apt/trusted.gpg.d/insynchq.gpg] http: //apt.insync.io/mint wilma non-free contrib
  Active apt repos in: /etc/apt/sources.list.d/nordvpn.list
    1: deb https: //repo.nordvpn.com/deb/nordvpn/debian stable main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com xia main upstream import backport
    2: deb http: //mirror.uoregon.edu/ubuntu noble main restricted universe multiverse
    3: deb http: //mirror.uoregon.edu/ubuntu noble-updates main restricted universe multiverse
    4: deb http: //mirror.uoregon.edu/ubuntu noble-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ noble-security main restricted universe multiverse
  Active apt repos in: /etc/apt/sources.list.d/signal-xenial.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/signal-desktop-keyring.gpg] https: //updates.signal.org/desktop/apt xenial main
Info:
  Memory: total: 28 GiB available: 27.33 GiB used: 3.66 GiB (13.4%)
  Processes: 429 Power: uptime: 7d 5h 54m states: freeze,mem,disk suspend: deep wakeups: 0
    hibernate: platform Init: systemd v: 255 target: graphical (5) default: graphical
  Compilers: gcc: 13.3.0 Client: Unknown python3.12 client inxi: 3.3.34

RP2, 2025-05-01:
Hey Brian, 

I experienced the same issue recently and stumbled upon [this forum post](https://forums.linuxmint.com/viewtopic.php?t=400535) which had a few helpful tips in it. One was running `dmesg` and looking for the bluetooth controller, in my case it listed the **firmware** as missing. 

Try to see if there's any firmware issues on your end, what fixed it for me was just rolling back the Linux kernel one revision as bluetooth was working just fine up until recently.

Hope this helps.

RP2, 2025-05-01:
Hey Bryan,

I experienced the same issue recently and stumbled upon [this forum post](https://forums.linuxmint.com/viewtopic.php?t=400535) which had a few helpful tips in it. One was running dmesg and looking for the bluetooth controller, in my case it listed the firmware as missing.

Try to see if there's any firmware issues on your end, what fixed it for me was just rolling back the Linux kernel one revision as bluetooth was working just fine up until recently.

Hope this helps.

RP2, 2025-05-01:
I experienced the same issue recently and stumbled upon [this forum post](https://forums.linuxmint.com/viewtopic.php?t=400535) which had a few helpful tips in it. One was running `dmesg` and looking for the bluetooth controller, in my case it listed the **firmware** as missing.

Try to see if there's any firmware issues on your end, what fixed it for me was just rolling back the Linux kernel one revision as bluetooth was working just fine up until recently.

Hope this helps.