blueman-project/blueman #2714: When booting it is always not ENABLED

Denisgsv, 2025-04-30:
Even if autoenable is true, every time after reboot bluetooth is disabled

Denisgsv, 2025-04-30:
dmesg | egrep -i 'blue|firm'
[    3.221514] Bluetooth: Core ver 2.22
[    3.226435] NET: Registered PF_BLUETOOTH protocol family
[    3.226437] Bluetooth: HCI device and connection manager initialized
[    3.226441] Bluetooth: HCI socket layer initialized
[    3.226442] Bluetooth: L2CAP socket layer initialized
[    3.226446] Bluetooth: SCO socket layer initialized
[    3.380776] mt7921e 0000:08:00.0: WM Firmware Version: ____000000, Build Time: 20240219103337
[    3.421986] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20240219103618
[    4.601203] [drm] Loading DMUB firmware via PSP: version=0x05000F00
[    4.601471] [drm] Found VCN firmware Version ENC: 1.30 DEC: 3 VEP: 0 Revision: 4
[    4.749580] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    4.749583] Bluetooth: BNEP filters: protocol multicast
[    4.749586] Bluetooth: BNEP socket layer initialized
[   24.189122] Bluetooth: hci0: Device setup in 20417603 usecs
[   24.189131] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[   24.465007] Bluetooth: hci0: AOSP extensions version v1.00
[   24.465014] Bluetooth: hci0: AOSP quality report is supported
[   24.486187] Bluetooth: MGMT ver 1.23
[   24.500394] Bluetooth: RFCOMM TTY layer initialized
[   24.500399] Bluetooth: RFCOMM socket layer initialized
[   24.500402] Bluetooth: RFCOMM ver 1.11
[   89.031031] Bluetooth: hci0: HW/SW Version: 0x008a008a, Build Time: 20240219103618
[   89.198153] Bluetooth: hci0: Device setup in 175279 usecs
[   89.198160] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[   89.478041] Bluetooth: hci0: AOSP extensions version v1.00
[   89.478047] Bluetooth: hci0: AOSP quality report is supported

RP2, 2025-05-01:
What do you get if you just enter `dmesg`? If it mentions anything about missing firmware, that was my issue. You can either reinstall the firmware of just roll back the Linux kernel one revision, which is what I did to fix it.

Denisgsv, 2025-05-02:
Hello, i managed to to solve my issue by "disabling Killswitch plugin" actually i am wondering what did it do exactly, why disabling it fixed my problem ? Was it actually fixed or its a workaround.
---------------------------------------------------------------------------------------------------------------------------

on dmesg i have an error 

```
[   24.724443] Bluetooth: RFCOMM TTY layer initialized
[   24.724448] Bluetooth: RFCOMM socket layer initialized
[   24.724451] Bluetooth: RFCOMM ver 1.11
[   25.941332] Bluetooth: hci0: ACL packet for unknown connection handle 3837
```

also wondering why is it there  ? its a fresh install so no tinkering has been done  so trying to understand what doesnt work why ?

RP2, 2025-05-02:
If it's a fresh install there's been [another kernel update](https://changelogs.ubuntu.com/changelogs/pool/main/l/linux/linux_6.8.0-59.61/changelog) which addressed a bluetooth CVE. I had the bluetooth issue only on 6.8.0-58, while 0-57 and 0-59 bluetooth works fine again. 

What kernel are you on now? Does toggling the plugin instantly break your bluetooth? Worst case you just leave _killswitch_ disabled for now.

Denisgsv, 2025-05-02:
> If it's a fresh install there's been [another kernel update](https://changelogs.ubuntu.com/changelogs/pool/main/l/linux/linux_6.8.0-59.61/changelog) which addressed a bluetooth CVE. I had the [...]

i was advised to update to update to 6.11 so thats what im using.  No for now it doesnt break anything i am using it with killswiitch of and for me works fine.

I just wonder what was killswitch supposed to do, and why it caused and fixed my issue when disabled. 

And maybe report this issue

cschramm, 2025-05-02:
I assume that if you exit or kill blueman-applet with Bluetooth disabled and start it again, it's still disabled.

Please exit or kill it and then run `blueman-applet --loglevel debug`, so that we can see what happens.

Denisgsv, 2025-05-02:
with killswitch off

`blueman-applet --loglevel debug
blueman-applet 21.53.46 INFO     PluginManager:86 load_plugin: ['GameControllerWakelock', 'StatusNotifierItem', 'ExitItem', 'NetUsage', 'TransferService', 'StandardItems', 'PulseAudioProfile', 'SerialManager', 'AutoConnect', 'KillSwitch', 'PowerManager', 'Networking', 'StatusIcon', 'DhcpClient', 'PPPSupport', 'DBusService', 'RecentConns', 'AuthAgent', 'Menu', 'ConnectionNotifier', 'DiscvManager', 'ShowConnected', 'NMPANSupport', 'NMDUNSupport', 'DisconnectItems']
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 21.53.46 WARNING  PluginManager:150 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.PulseAudioProfile.AudioProfiles'>
blueman-applet 21.53.46 INFO     PulseAudioUtils:207 pa_context_event: 1
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 21.53.46 WARNING  PluginManager:150 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 21.53.46 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 21.53.46 INFO     PulseAudioUtils:207 pa_context_event: 2
blueman-applet 21.53.46 INFO     TransferService:233 _on_dbus_name_appeared: org.bluez.obex :1.57
blueman-applet 21.53.46 INFO     PulseAudioUtils:207 pa_context_event: 3
blueman-applet 21.53.46 INFO     Applet:76 _on_dbus_name_appeared: org.bluez :1.4
blueman-applet 21.53.46 INFO     Functions:99 launch    : Gtk eventtime is 0, not using LaunchContext
blueman-applet 21.53.46 INFO     Networking:78 set_nap   : set nap False
blueman-applet 21.53.46 INFO     BluezAgent:56 register_agent: Register Agent
blueman-applet 21.53.46 INFO     AgentManager:17 on_registered: /org/bluez/obex/agent/blueman
blueman-applet 21.53.46 DEBUG    Base:67 _properties_changed: /org/bluez/hci0 {'Pairable': True} [] <Adapter.Adapter object at 0x705e50fdcd40 (blueman+bluez+Adapter+Adapter at 0x18dc6a40)>
blueman-applet 21.53.46 DEBUG    DiscvManager:95 on_adapter_property_changed: prop Pairable True
blueman-applet 21.53.46 INFO     ShowConnected:55 enumerate_connections: Found 1 existing connections
blueman-applet 21.53.46 INFO     PulseAudioUtils:207 pa_context_event: 4
blueman-applet 21.53.46 INFO     PulseAudioProfile:135 on_pa_ready: PulseAudio Ready
blueman-applet 21.53.46 INFO     PulseAudioUtils:213 <lambda>  : 1
blueman-applet 21.54.11 INFO     PulseAudioUtils:337 __event_callback: 25 89
blueman-applet 21.54.11 DEBUG    PulseAudioProfile:111 on_pa_event: 25 89
blueman-applet 21.54.11 INFO     PulseAudioProfile:124 on_pa_event: card
blueman-applet 21.54.11 INFO     PulseAudioProfile:126 on_pa_event: change
`

with killswitch on

`blueman-applet --loglevel debug
[... cut ...]
blueman-applet 21.55.26 INFO     PulseAudioUtils:207 pa_context_event: 4
blueman-applet 21.55.26 INFO     PulseAudioProfile:135 on_pa_ready: PulseAudio Ready
blueman-applet 21.55.26 INFO     PulseAudioUtils:213 <lambda>  : 1
blueman-applet 21.55.29 INFO     PulseAudioUtils:337 __event_callback: 25 89
blueman-applet 21.55.29 DEBUG    PulseAudioProfile:111 on_pa_event: 25 89
blueman-applet 21.55.29 INFO     PulseAudioProfile:124 on_pa_event: card
blueman-applet 21.55.29 INFO     PulseAudioProfile:126 on_pa_event: change
`

cschramm, 2025-05-02:
Not sure I can follow 100 %, but I guess you need to...

1. Enable the plugin
2. Reboot
3. See Bluetooth disabled
4. Stop blueman-applet
5. Run `blueman-applet --loglevel debug`
6. Still see Bluetooth disabled

If point 6 does not hold, you might need to disable blueman's autostart in order to produce a proper log after booting.