blueman-project/blueman #2724: Bug, Suggestion: Blueman should not start newly connected bluetooth device with 100% volume

AmirrezaFiroozi, 2025-05-06:
**TLDR:** Don't set volume=100% on first bluetooth audio device connection

When connecting to a bluetooth headset/earbuds for the first time,
Blueman sets the volume to 100%.

If the user is not careful when they play media, the volume could be dangerously high,
And it could 1. Be quite uncomfortable 2. Potentially damage the person's hearing.

**Steps to reproduce:**
1. Unpair an earphone/earbuds/headset/loudspeaker
2. Pair and connect
3. Press the volume down key (Observe the volume decreases from 100%)

**Additional info: (might be helpful)**
OS = Archlinux
Ver = extra/blueman 2.4.4-1

Earbuds = Soundcore R50i (is this relevant?)

Sound related packages:
pipewire 1:1.4.2-1
pipewire-audio 1:1.4.2-1
wireplumber 0.5.8-1

**Potential Solution:**
After the connection is successful and the output is connected to an audio sink,
Set a predetermined volume (30% or 40%? This is up for debate. The clear thing is that it must not be 100%!)

cschramm, 2025-05-06:
Do you have a reason to believe that blueman does so? PipeWire is handling all the audio aspects for you. blueman exposes an interface to switch the profile. That's all.

AmirrezaFiroozi, 2025-05-06:
Okay then. I will report this bug on PipeWire development repo.

closed 2025-05-06 by cschramm