blueman-project/blueman #2765: Tray icon fails to appear on session start

Vladimir-csp, 2025-06-21:
Debian sid, Hyprland, waybar, blueman-applet 2.4.4
When XDG autostart entry starts as `app-blueman@autostart.service` service at session start, icon does not appear. Service logs warning:
```
blueman-tray[2831475]: gtk_widget_get_scale_factor: assertion 'GTK_IS_WIDGET (widget)' failed
```
But continues running. Icon only appears after restarting service. Perhaps it does not wait for tray to become available and just gives up on showing icon.

cschramm, 2025-06-23:
It looks like the logic is broken. I'm puzzled if and how #4334 ever worked, as the message is actually prefixed with `GDBus.Error:`.

Can you try the following change?

```diff
--- a/blueman/main/indicators/StatusNotifierItem.py
+++ b/blueman/main/indicators/StatusNotifierItem.py
@@ -155,7 +155,7 @@ class StatusNotifierItem(IndicatorInterface):
                 None, Gio.DBusCallFlags.NONE, -1)
             watcher_expected = True
         except GLib.Error as e:
-            watcher_expected = not e.message.startswith("org.freedesktop.DBusError.ServiceUnknown")
+            watcher_expected = not e.message.startswith("GDBus.Error:org.freedesktop.DBusError.ServiceUnknown")
             raise IndicatorNotAvailable
 
     def set_icon(self, icon_name: str) -> None:
```

You can just do that manually in the installed file.

Vladimir-csp, 2025-06-23:
its `GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown` (a dot between `DBus` and `Error`). Yes, it fixes the problem.

PR #2779 (2025-07-01): Fix handling for new StatusNotifierWatcher
blueman-tray restarts when a new StatusNotifierWatcher appears. This is especially relevant when blueman and a watcher like waybar get started in the session in no particular order. Unfortunately we have to use the `GLib.Error.message` and it looks like #2334 lacked the prefix; at least I cannot find any hint that this could have worked.

Fixes #2765
commit 67209a43 (2025-07-01): Fix handling for new StatusNotifierWatcher (#2779)

blueman-tray restarts when a new StatusNotifierWatcher appears. This is especially relevant when blueman and a watcher like waybar get started in the session in no particular order. Unfortunately we have to use the `GLib.Error.message` and it looks like #2334 lacked the prefix; at least I cannot find any hint that this could have worked.

Fixes #2765
commit c15a86e9 (2025-07-01): Fix handling for new StatusNotifierWatcher (#2779)

blueman-tray restarts when a new StatusNotifierWatcher appears. This is especially relevant when blueman and a watcher like waybar get started in the session in no particular order. Unfortunately we have to use the `GLib.Error.message` and it looks like #2334 lacked the prefix; at least I cannot find any hint that this could have worked.

Fixes #2765
closed 2025-07-01 by cschramm