blueman-project/blueman #2769: Adapters Agent Problem

Leo2305, 2025-06-21:
I've had some problems with Blueman and it does not make my bluetooth work:

This is what I got from running blueman-manager from terminal:
```Traceback(most recent call last):
      File "/usr/lib/python3/dist-packages/blueman/main/Manager.py", line 159, in_on_bt_state_changed
          self.Applet.SetBluetoothStatus("(b)", state)
          ^ ^ ^ ^ ^ ^
AttributeError: 'Blueman' object has no attribute 'Applet'```

Leo2305, 2025-06-22:
So, basically, I found a racing threads problem that, somehow, consistently wins every time I run blueman. It would be necessary to make an if statement check to make sure `Applet` gets set before running `self.Applet.SetBluetoothStatus("(b)", state)` ? Just currious.

Leo2305, 2025-06-22:
blueman-manager 13.33.17 INFO     PluginManager:86 load_plugin: ['Notes', 'Info', 'PulseAudioProfile', 'Services']
blueman-manager 13.33.17 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.manager.Notes.Notes'>
blueman-manager 13.33.17 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.manager.Info.Info'>
blueman-manager 13.33.17 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'>
blueman-manager 13.33.17 INFO     PulseAudioUtils:207 pa_context_event: 1
blueman-manager 13.33.17 INFO     PluginManager:159 __load_plugin: loading <class 'blueman.plugins.manager.Services.Services'>

(blueman-manager:8480): Gtk-WARNING **: 13:33:17.488: Theme directory 16/16 of theme Infinity has no size field

(blueman-manager:8480): Gtk-WARNING **: 13:33:17.490: Theme directory actions/apps of theme Infinity has no size field

(blueman-manager:8480): Gtk-WARNING **: 13:33:17.490: Theme directory 16/categories of theme Infinity has no size field

(blueman-manager:8480): Gtk-WARNING **: 13:33:17.490: Theme directory 16/devices of theme Infinity has no size field

(blueman-manager:8480): Gtk-WARNING **: 13:33:17.490: Theme directory 16/emblems of theme Infinity has no size field

(blueman-manager:8480): Gtk-WARNING **: 13:33:17.490: Theme directory 16/emotes of theme Infinity has no size field

(blueman-manager:8480): Gtk-WARNING **: 13:33:17.490: Theme directory 16/legacy of theme Infinity has no size field

(blueman-manager:8480): Gtk-WARNING **: 13:33:17.490: Theme directory 16/mimetypes of theme Infinity has no size field

(blueman-manager:8480): Gtk-WARNING **: 13:33:17.490: Theme directory 16/places of theme Infinity has no size field

(blueman-manager:8480): Gtk-WARNING **: 13:33:17.490: Theme directory 16/status of theme Infinity has no size field

(blueman-manager:8480): Gtk-WARNING **: 13:33:17.490: Theme directory 16/256 of theme Infinity has no size field

(blueman-manager:8480): Gtk-WARNING **: 13:33:17.490: Theme directory categories/16 of theme Infinity has no size field

(blueman-manager:8480): Gtk-WARNING **: 13:33:17.490: Theme directory devices/32 of theme Infinity has no size field

(blueman-manager:8480): Gtk-WARNING **: 13:33:17.490: Theme directory devices/8 of theme Infinity has no size field

(blueman-manager:8480): Gtk-WARNING **: 13:33:17.492: Theme directory status/ui of theme Infinity has no size field

blueman-manager 13.33.17 INFO     PulseAudioUtils:207 pa_context_event: 2
blueman-manager 13.33.17 INFO     PulseAudioUtils:207 pa_context_event: 3
blueman-manager 13.33.17 INFO     PulseAudioUtils:207 pa_context_event: 4
blueman-manager 13.33.17 INFO     PulseAudioProfile:32 on_pa_ready: connected
blueman-manager 13.33.17 INFO     PulseAudioUtils:213 <lambda>  : 1
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/main/Manager.py", line 159, in _on_bt_state_changed
    self.Applet.SetBluetoothStatus("(b)", state)
    ^^^^^^^^^^^
AttributeError: 'Blueman' object has no attribute 'Applet'
blueman-manager 13.33.42 INFO     Manager:98 on_dbus_name_vanished: org.bluez
blueman-manager 13.33.44 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.manager.Services.Services object at 0x7168c33a5cd0>
blueman-manager 13.33.44 INFO     PulseAudioUtils:380 _on_delete: Destroying PulseAudioUtils instance
blueman-manager 13.33.44 INFO     PulseAudioUtils:207 pa_context_event: 6
blueman-manager 13.33.44 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile object at 0x7168c33a7170>
blueman-manager 13.33.44 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.manager.Info.Info object at 0x7168c6551bb0>
blueman-manager 13.33.44 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.manager.Notes.Notes object at 0x7168c7134c20>

infirit, 2025-06-22:
It looks like the bluetooth service isn't running and blueman-manager goes through `on_dbus_name_vanished`. You should see a a dialog saying the bluez daemon isn't running?

`on_dbus_name_vanished` does bypass the `Applet` creation because there is no point doing so as we're going to exit soon anyway. The error makes a little more  sense now but I still don't see how you end up in `_on_bt_state_changed` in this code path 🤔 .

Leo2305, 2025-06-22:
Yeah the Bluez daemon not running message appears, but I have no idea why that happens. Could it be a startup `systemctl` problem?

Leo2305, 2025-06-22:
**UPDATE: I FOUND THE SOLUTION!!!**
[solution here](https://www.reddit.com/r/linuxmint/comments/126yoob/bluetooth_connectivity_died/)
Apparently I needed to unistall and reinstall the entire bluetooth aparatus and fully shutdown the computer for it to stick.
What exactly was giving it that problem? I don't think it was blueman, but something just kills the `bluetooth.service` and you have to reset it. It's better to check it on your end and if it is an option that is not compatible with some hardware. My motherboard is a ROG Strix B-550 Wifi II, so check the bluetooth chip in there.