blueman-project/blueman #2770: Crash on LE headset connection attempt – KeyError in ManagerDeviceMenu (Fedora 40 + Hyprland)

rs0125, 2025-06-22:
**Title:** Crash on LE headset connection attempt – `KeyError` in `ManagerDeviceMenu` (Fedora 40 + Hyprland)

**Description:**
While using **Blueman on Fedora 40 with Hyprland**, I encountered an exception when attempting to connect my **Bluetooth headset** (LE device). The connection fails and `blueman-manager` throws a traceback, ending in a `KeyError`.

This occurred with `blueman-manager` version: **\[insert exact version if known or output of `blueman-manager --version`]**.

**Steps to Reproduce:**

1. Launch `blueman-manager`.
2. Attempt to connect to a Bluetooth LE headset.
3. Connection fails with the following traceback.

**Error Logs:**

```
blueman-manager 14.14.07 WARNING  ManagerDeviceMenu:145 fail      : fail g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.Failed: blueman.bluez.errors.DBusInProgressError:  br-connection-busy (0)
Traceback (most recent call last):
  File "/usr/lib64/python3.13/site-packages/gi/overrides/Gio.py", line 314, in __async_result_handler
    ret = obj.call_finish(result)
gi.repository.GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.Failed: blueman.bluez.errors.DBusFailedError:  le-connection-abort-by-local (0)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib64/python3.13/site-packages/gi/overrides/Gio.py", line 319, in __async_result_handler
    error_callback(obj, e, real_user_data)
  File "/usr/lib/python3.13/site-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 144, in fail
    self.unset_op(device)
  File "/usr/lib/python3.13/site-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 121, in unset_op
    del ManagerDeviceMenu.__ops__[device.get_object_path()]
KeyError: '/org/bluez/hci0/dev_C7_86_42_00_5F_9D'
```

**Expected Behavior:**
Connection should fail gracefully without raising a Python exception or crashing the UI.

**System Info:**

* OS: Fedora 40 (x86\_64)
* DE: Hyprland
* Blueman: blueman-2.4.4-2.fc41.x86_64
* Python: 3.13
* bluetoothctl: 5.83

**Additional Notes:**

* The issue seems to stem from an internal state mismatch in `ManagerDeviceMenu.__ops__`, likely due to the async call failing mid-connection.
* I'm using PipeWire as the audio backend (if relevant).

infirit, 2025-06-22:
`le-connection-abort-by-local` essentially means the headset is not responding to the connection request. There isn't much we can do about that. You can try to remove the device from the list, search for it and try to connect.

If it persists post the complete debug log here and maybe we can see what it going on. From a terminal `blueman-manager --loglevel debug` and post the terminal output here.

rs0125, 2025-06-22:
Hi @infirit , thanks for the quick response.

I understand that `le-connection-abort-by-local` typically points to a device-side issue, but my concern here is not about the connection failure itself — it's about how Blueman handles that failure.

The issue is that Blueman raises an **uncaught `KeyError`** resulting in a raw Python stack trace in the UI, which indicates that the failure isn't being handled gracefully within the code. Specifically, this line crashes:

```python
del ManagerDeviceMenu.__ops__[device.get_object_path()]
```

This suggests that `device.get_object_path()` doesn't exist in `__ops__` at that point, possibly due to an earlier failure or race condition. Regardless of the device's behavior, the UI shouldn't be throwing an exception like this to the user, and could handle the exception more gracefully with a try-catch, considering its a pretty common failure.

Let me know if you'd like me to run with `--loglevel debug` and post the full trace — happy to help further debug this, and I could possibly help with a patch too.

infirit, 2025-06-22:
yeah, do post a log as `connect_service` which is called to connect the device will call `set_op` which add adds to the dict.

You should see `set_op` twice in the log, see below for an example.

```
blueman-manager 21.30.54 DEBUG    ManagerDeviceMenu:307 generate  : 8BitDo Ultimate 2 Wireless
blueman-manager 21.30.55 INFO     ManagerDeviceMenu:110 set_op    : op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f58e73a41c0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x55e310d97000)>
blueman-manager 21.30.55 INFO     ManagerDeviceMenu:110 set_op    : op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f58e73b1e40 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x55e310ed0c90)>
blueman-manager 21.31.35 INFO     ManagerDeviceMenu:123 unset_op  : op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f58e73a41c0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x55e310d97000)>
blueman-manager 21.31.35 DEBUG    ManagerDeviceMenu:307 generate  : 8BitDo Ultimate 2 Wireless
blueman-manager 21.31.35 INFO     ManagerDeviceMenu:123 unset_op  : op: regenerating instance <ManagerDeviceMenu.ManagerDeviceMenu object at 0x7f58e73b1e40 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x55e310ed0c90)>
blueman-manager 21.31.35 WARNING  ManagerDeviceMenu:145 fail      : fail g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.Failed: blueman.bluez.errors.DBusFailedError:  le-connection-abort-by-local
```