blueman-project/blueman #2791: Bluetooth Connection Issue - Linux Mint

amerucandior, 2025-07-11:
**Problem:**
Bluetooth device pairs successfully on initial connection but fails to reconnect after disconnection. Connection only works after full system restart + device unpairing/re-pairing cycle.
Environment:

**OS: Linux Mint**
Issue: Bluetooth device connection state management
Symptom: Device appears paired but connection attempts fail after first disconnect

**Current Workaround:**
bash# Current manual process required for reconnection:
1. Restart system
2. Remove device from paired devices
3. Re-pair device
Expected Behavior:
Device should reconnect automatically or allow manual reconnection without requiring system restart and re-pairing.
Troubleshooting Steps Needed:

Check bluetoothctl logs during failed reconnection attempts
Verify bluetooth service status and restart behavior
Examine device power management settings
Check for pulseaudio/pipewire bluetooth module issues

Logs to Gather:
bashsudo journalctl -u bluetooth
bluetoothctl show
dmesg | grep -i bluetooth

amerucandior, 2025-07-11:
dmesg | grep -i bluetooth

[    3.858952] thinkpad_acpi: rfkill switch tpacpi_bluetooth_sw: radio is unblocked
[    4.855524] Bluetooth: Core ver 2.22
[    4.865349] NET: Registered PF_BLUETOOTH protocol family
[    4.865356] Bluetooth: HCI device and connection manager initialized
[    4.865362] Bluetooth: HCI socket layer initialized
[    4.865365] Bluetooth: L2CAP socket layer initialized
[    4.865375] Bluetooth: SCO socket layer initialized
[    5.011170] Bluetooth: hci0: BCM: chip id 63
[    5.012337] Bluetooth: hci0: BCM: features 0x07
[    5.028258] Bluetooth: hci0: BCM20702A
[    5.028267] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[    5.030494] Bluetooth: hci0: BCM: firmware Patch file not found, tried:
[    5.030501] Bluetooth: hci0: BCM: 'brcm/BCM20702A1-0a5c-21e6.hcd'
[    5.030503] Bluetooth: hci0: BCM: 'brcm/BCM-0a5c-21e6.hcd'
[    5.353728] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    5.353737] Bluetooth: BNEP filters: protocol multicast
[    5.353745] Bluetooth: BNEP socket layer initialized
[    5.367891] Bluetooth: MGMT ver 1.22
[   10.278636] Bluetooth: RFCOMM TTY layer initialized
[   10.278657] Bluetooth: RFCOMM socket layer initialized
[   10.278668] Bluetooth: RFCOMM ver 1.11
[  849.486935] Bluetooth: hci0: command tx timeout
[  851.534893] Bluetooth: hci0: command tx timeout
[  853.582822] Bluetooth: hci0: command tx timeout
[  855.630765] Bluetooth: hci0: command tx timeout
[  857.678650] Bluetooth: hci0: command tx timeout
[  858.315953] Bluetooth: hci0: link tx timeout
[  858.315966] Bluetooth: hci0: killing stalled connection 41:42:03:55:80:ae
[  859.726600] Bluetooth: hci0: command tx timeout
[  861.774538] Bluetooth: hci0: command 0x1405 tx timeout
[  863.822365] Bluetooth: hci0: command 0x1405 tx timeout
[  865.870270] Bluetooth: hci0: command 0x1405 tx timeout
[  867.918242] Bluetooth: hci0: command 0x1405 tx timeout
[  869.966169] Bluetooth: hci0: command 0x1405 tx timeout
[  872.014078] Bluetooth: hci0: command 0x1405 tx timeout
[  874.061934] Bluetooth: hci0: command 0x1405 tx timeout
[  876.110129] Bluetooth: hci0: command 0x1405 tx timeout
[  878.157871] Bluetooth: hci0: command 0x1405 tx timeout
[  912.460579] Bluetooth: hci0: command 0x1405 tx timeout
[  917.004255] Bluetooth: hci0: command 0x1405 tx timeout
[  957.515052] Bluetooth: hci0: command 0x1405 tx timeout
[ 7836.380396] Bluetooth: hci0: command 0x1405 tx timeout
[ 7879.513983] Bluetooth: hci0: command 0x1405 tx timeout
[ 7881.625907] Bluetooth: hci0: command 0x1405 tx timeout
[ 7909.400954] Bluetooth: hci0: command 0x1405 tx timeout
[ 7925.016422] Bluetooth: hci0: Opcode 0x2005 failed: -110
[ 7925.016422] Bluetooth: hci0: command 0x1405 tx timeout
[ 7925.016451] Bluetooth: hci0: Opcode 0x200b failed: -110
[ 7927.065443] Bluetooth: hci0: command 0x1405 tx timeout

alfredcurl, 2025-07-12:
I have the same issue

Grumpster-Dev, 2025-07-12:
I have a very similar issue.  My Logitech Ergo K860 keyboard worked briefly (but intermittently) this morning on linux-mint, then I noticed that several updates for linux where available, and I updated and restarted my machine.  Now when I attempt to pair the keyboard, it sees it.  Pairs it, then disconnects it immediately and reports Pairing Failed for Ergo K860 followed by some hex code or similar.  Have updated bluetooth via terminal, restarted, repaired...Nada.  Keyboard works just fine with my macOS machine.  This just started today...the keyboard had been working on this install of linux for several weeks prior.